Optiboot/optiboot #283: LED options ambiguity.

dfxhub, 2020-04-18:
Hi!

There are ambiguity in LED behaviour when mixing different compile options in the latest version.

For example:
LED_START_FLASHES=0 LED_DATA_FLASH=1 LED_START_ON=1 - LED is flashing only during data transfer, all other time it stays off (include startup time).

I think this is because of getch() code. If LED_DATA_FLASH=1 at [#710](https://github.com/Optiboot/optiboot/blob/a3470cc0c5d9e13b284499e2f7c96c5c8a46764b/optiboot/bootloaders/optiboot/optiboot.c#L710) it turns on and then immidiatly turns off at the very beginning of getch() - [#1014](https://github.com/Optiboot/optiboot/blob/a3470cc0c5d9e13b284499e2f7c96c5c8a46764b/optiboot/bootloaders/optiboot/optiboot.c#L1014) LED_PIN |= _BV(LED); And if there are no serial data LED stays off forever. All is working logically if I comment out lines [#1007](https://github.com/Optiboot/optiboot/blob/a3470cc0c5d9e13b284499e2f7c96c5c8a46764b/optiboot/bootloaders/optiboot/optiboot.c#L1007) to [#1016](https://github.com/Optiboot/optiboot/blob/a3470cc0c5d9e13b284499e2f7c96c5c8a46764b/optiboot/bootloaders/optiboot/optiboot.c#L1016). LED turns on during boot and stays on until serial data is coming ([#1072](https://github.com/Optiboot/optiboot/blob/a3470cc0c5d9e13b284499e2f7c96c5c8a46764b/optiboot/bootloaders/optiboot/optiboot.c#L1072) at the end of getch().

Regards.

WestfW, 2020-04-18:
LED_DATA_FLASH ought not interfere with LED_START_FLASHES, because  the whole `flash_led(LED_START_FLASHES*2)` call should have completed before the first call to `getch()`

dfxhub, 2020-04-19:
Sure. But situation which I described is a bit confusing. I expected to decrease code size by disabling LED flashing code but without losing the ability to identify that the bootloader is starting by defining LED_START_ON. As a result I've got blank LED upon startup. I checked connections, reflushed over and over again until realized that something wrong with the code. ;)

WestfW, 2020-05-11:
Ah.  LED_DATA_FLASHES conflicts with LED_START_ON; that makes more sense.
Although I don't know whether it makes sense to use them together.  (I'm not certainly that LED_DATA_FLASHES makes sense at all, as implemented.)