Optiboot/optiboot #286: How up to USART 2M ?

selevo, 2020-05-11:
Error 0% for 16MHz

WestfW, 2020-05-11:
"make <target> BAUD_RATE=2000000" ?
I don't know if it works at that rate (you'd have fewer than 80 instructions per character to process it), but the MCUDude cores have pre-built bootloaders for up to 1Mbps.
IIRC, people who have tried it have found 2Mbps not to be any faster than 1Mbps, overall.  Actual flash write time becomes the dominating factor (~4ms per 128byte page on a m328p - about 320kbps.)

Or perhaps I didn't understand the question?

selevo, 2020-05-12:
Thank you, everyone understood correctly.
I somehow did not think about flash memory speed.
So you can raise the speed in two? (BAUDE 230400)

WestfW, 2020-05-22:
Can your host do 256000bps?  That would be zero-error. 230400 has a 3.5% error, which is not good...
Users have reported apparently increased performance running up to 1Mbps.  2Mbps (the absolute max that the chip is capable of) does not seem worthwhile.

closed 2020-05-22 by WestfW