Optiboot/optiboot #287: how to modify optboot to accept rs485 serial in atmega 4809

NeyCsn, 2020-05-21:
Hello people. i need to implement rs485 in optbootx for atmega 4809. however i was not successful following the steps indicated here in optboot.
thank you

WestfW, 2020-05-22:
Could you be more specific?  Optiboot does not currently support rs485 at all.
Getting it to work would be dependent on which transceiver you are using and how you have it connected, and what software you intend to drive it with.  And maybe on avrdude as well.
Did you see https://github.com/SodaqMoja/optiboot ?

NeyCsn, 2020-05-22:
my doubt is very simple. how I get the files "Optboot_x.c", "makefile", "Boot_opt.h", ... and generate a new .hex and .elf file.

WestfW, 2020-05-23:
`make -f Makefile.mega0 atmega4809 LED=D6 UARTTX=A0`

WestfW, 2020-05-30:
I noticed that "make atmega4809" results in compile errors rather than "unknown target" errors, from a time when I was thinking of having a common makefile, and I've removed those targets.
This should help avoid confusion.

closed 2020-05-30 by WestfW