Optiboot/optiboot #288: How to turn built in led completly off?

FrisscoCBR, 2020-06-17:
So i am using a command like this to create a bootloader to flash on to atmega328p so i dont have built in led on never. 

omake LED_START_FLASHES=0 LED_START_ON=0 atmega328

Everything goes as expected. It uploads as expected but then built in LED turns full on until i upload a blank sketch. After that whe i hit RESET on arudino in again turns on led for a short time and than nothing as expected. 

The thing is i want it to never turn on. To not pull that pin HIGH ever, untill i tell it to. Its really important becuase i am controlling a cnc spindle VFD with that pin,and i can not use any other pin becuase GRBL is defined in that way!

Thanks in advance!

WestfW, 2020-06-18:
Fixed in https://github.com/Optiboot/optiboot/commit/6f98751c6d184ec4a9c7f1fe0f0dcf12f9db2cc9
All of the option symbols are now 0/1 instead of sometimes defined/undefined.

closed 2020-06-18 by WestfW

FrisscoCBR, 2020-06-18:
After testing out your new code and using the same command 

omake LED_START_FLASHES=0 LED_START_ON=0 atmega328

and burn it to the chip now after i restart the power to the arduino and upload the blank sketch built in led stays fully on! I can record the process if you need it! I am using arduino uno with atmega328P-PU. Thanks