Optiboot/optiboot #295: Optiboot compile problem

mmzeynalli, 2020-10-12:
Good day,
I am currently trying to write my own bootloader, using optiboot. I faced problem using make, and I thought it was error from my piece of code. After that, I decided to try with original, unchanged form. However, I faced the same problem:

`$ make atmega328
avr-gcc (WinAVR 20100110) 4.3.3
Copyright (C) 2008 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3              -c -o optiboot.o optiboot.c
optiboot.c:367:6: error: #error BAUD_RATE off by greater than -5%
optiboot.c:385:2: error: #error Unachievable baud rate (too fast) BAUD_RATE
optiboot.c:444: warning: 'leaf' attribute directive ignored
optiboot.c:445: warning: 'leaf' attribute directive ignored
optiboot.c: In function 'main':
optiboot.c:701: error: 'BAUD_SETTING' undeclared (first use in this function)
optiboot.c:701: error: (Each undeclared identifier is reported only once
optiboot.c:701: error: for each function it appears in.)
make: *** [optiboot.o] Error 1`

I don't get why I have baud problem (it is being too fast), but still, I have undefined BAUD_SETTING. Should I declare it manually in the start of optiboot.c file?

P. S. Just to be sure, I am doing things right, I should `make` changed version of optiboot, which will create hex file, which in its turn will be able to be burned to MCU. Am I right?

mmzeynalli, 2020-10-12:
Just to clarify, I just successfully exported hex and lst files of modified version of optiboot v4 (officially Arduino supported)

WestfW, 2020-10-13:
1. "_I should make changed version of optiboot, which will create hex file, which in its turn will be able to be burned to MCU._"
   Yes, that's right.

2. You'll need a newer compiler than WINAVR.  Use the version that comes with current Arduino. or from Microchip: https://www.microchip.com/mplab/avr-support/avr-and-arm-toolchains-c-compilers
3. I'm not sure why you're getting the Baud Rate error.  When I compile here with an old compiler, it doesn't do that...

~~~
 make GCCROOT=/usr/local/CrossPack-AVR/bin/ atmega328
CrossPack-AVR/          CrossPack-AVR-20121207/ 
CrossPack-AVR-20100115/ CrossPack-AVR-48/       
WWHackintosh<5431> make GCCROOT=/usr/local/CrossPack-AVR-20100115/bin/ atmega328
avr-gcc (GCC) 4.3.3
Copyright (C) 2008 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
/usr/local/CrossPack-AVR-20100115/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3              -c -o optiboot.o optiboot.c
optiboot.c:386:6: warning: #warning BAUD_RATE off by greater than 2%
optiboot.c:455: warning: 'leaf' attribute directive ignored
optiboot.c:456: warning: 'leaf' attribute directive ignored
/usr/local/CrossPack-AVR-20100115/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3            -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o 
/usr/local/CrossPack-AVR-20100115/lib/gcc/avr/4.3.3/../../../../avr/bin/ld: section .version [00007ffe -> 00007fff] overlaps section .text [00007e00 -> 00008003]
make: *** [optiboot_atmega328.elf] Error 1
~~~

mmzeynalli, 2020-10-13:
Thanks for the answer.

I uninstalled avr-gcc, and used `export ENV=arduino`. After that, it got better:

```
$ make atmega328
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
../../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3              -c -o optiboot.o optiboot.c
optiboot.c:386:6: warning: #warning BAUD_RATE off by greater than 2% [-Wcpp]
     #warning BAUD_RATE off by greater than 2%
      ^
../../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3            -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o
../../../../tools/avr/bin/avr-size optiboot_atmega328.elf
   text    data     bss     dec     hex filename
    482       0       0     482     1e2 optiboot_atmega328.elf
../../../../tools/avr/bin/avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
../../../../tools/avr/bin/avr-objdump -h -S optiboot_atmega328.elf > optiboot_atmega328.lst
rm optiboot.o
```

It did not give error (and it generated .hex and .lst), but, it shows baud difference of 2.1%. Is it normal?

WestfW, 2020-10-15:
>> it shows baud difference of 2.1%. Is it normal?
Yes.
Fortunately 2.1% is within permissible error range (especially if the other side has high accuracy.)
In the case of an Arduino Uno, where the USB/Serial interface is ALSO an AVR with the same BRG behavior, the two sides end up matching "exactly" (assuming accurate clocks.)

closed 2021-04-21 by WestfW