Optiboot/optiboot #300: EEPROM support does not fit in 512 bytes

SukkoPera, 2020-11-09:
I'm trying to compile an Optiboot binary that has EEPROM support and fits in 512 bytes. I have tried different versions of the compiler, but I always get some variant of the following error:
```
$ make atmega328 LED=B5 UART=0 BAUD_RATE=38400 AVR_FREQ=8000000 SUPPORT_EEPROM=1 LED_START_FLASHES=0 NO_APP_SPM=1 SINGLESPEED=1
avr-gcc (GCC) 4.3.2
Copyright (C) 2008 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 38400, Real: 38461, UBRRL = 12, Difference=0.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=8000000  -DBAUD_RATE=38400 -DLED_START_FLASHES=0    -DLED=B5 -DSINGLESPEED=1 -DSUPPORT_EEPROM  -DAPP_NOSPM=1     -c -o optiboot.o optiboot.c
optiboot.c:411: warning: ‘leaf’ attribute directive ignored
optiboot.c:412: warning: ‘leaf’ attribute directive ignored
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=8000000  -DBAUD_RATE=38400 -DLED_START_FLASHES=0    -DLED=B5 -DSINGLESPEED=1 -DSUPPORT_EEPROM  -DAPP_NOSPM=1   -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o 
/opt/arduino-1.0.6/hardware/tools/avr/bin/../lib/gcc/avr/4.3.2/../../../avr/bin/ld: section .version [0000000000007ffe -> 0000000000007fff] overlaps section .text [0000000000007e00 -> 0000000000008017]
make: *** [Makefile:667: optiboot_atmega328.elf] Error 1
```

I guess this means the bootloader has grown too big, as I can get it to build if I add BIGBOOT=1, but the documentation seems to imply that what I want to achieve is possible.

Any hints?

PS: Actually I would like to build a version for the ATmega88p, but that seems even harder to do, so I thought I'd first try to make one for the 328p.

WestfW, 2020-11-10:
It looks like it works with gcc 5.4 (ie latest version actually shipped from Atmel.
https://www.microchip.com/mplab/avr-support/avr-and-arm-toolchains-c-compilers

It definitely doesn't work with 4.3.2 ("WinAVR"), and it doesn't seem to work with the 7.3 version that Arduino is shipping with the IDE (which I suppose I should investigate.)

SukkoPera, 2020-11-10:
I've managed to build it for the 88p (not tested yet) with the GCC from the latest Arduino (7.3.0) and the very same command line I used above:

```
$ make atmega88pa LED=B5 UART=0 BAUD_RATE=38400 AVR_FREQ=8000000 SUPPORT_EEPROM=1 LED_START_FLASHES=0 NO_APP_SPM=1 SINGLESPEED=1
avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 38400, Real: 38461, UBRRL = 12, Difference=0.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega88p -DF_CPU=8000000  -DBAUD_RATE=38400 -DLED_START_FLASHES=0    -DLED=B5 -DSINGLESPEED=1 -DSUPPORT_EEPROM  -DAPP_NOSPM=1   -DUART=0   -c -o optiboot.o optiboot.c
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega88p -DF_CPU=8000000  -DBAUD_RATE=38400 -DLED_START_FLASHES=0    -DLED=B5 -DSINGLESPEED=1 -DSUPPORT_EEPROM  -DAPP_NOSPM=1   -DUART=0 -Wl,--section-start=.text=0x1e00 -Wl,--section-start=.version=0x1ffe -Wl,--gc-sections -Wl,--undefined=optiboot_version -Wl,--relax -nostartfiles -o optiboot_atmega88p_UART0_38400_8000000.elf optiboot.o 
avr-size optiboot_atmega88p_UART0_38400_8000000.elf
   text	   data	    bss	    dec	    hex	filename
    512	      0	      0	    512	    200	optiboot_atmega88p_UART0_38400_8000000.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega88p_UART0_38400_8000000.elf optiboot_atmega88p_UART0_38400_8000000.hex
avr-objdump -h -S optiboot_atmega88p_UART0_38400_8000000.elf > optiboot_atmega88p_UART0_38400_8000000.lst
rm optiboot.o
```

I made a lot of attempts before opening this issue, but I must have missed this particular combination of options, sorry.

BTW, I went as back as GCC 4.3 because [the docs](https://github.com/Optiboot/optiboot/blob/master/Wiki/CompilingOptiboot.md) say so.

SukkoPera, 2020-11-10:
Tested, seems to be working well for both flash and eeprom.

HEX file attached in case anyone needs it.

Thanks for the prompt reply :).

[optiboot_atmega88p_UART0_38400_8000000.hex.txt](https://github.com/Optiboot/optiboot/files/5515611/optiboot_atmega88p_UART0_38400_8000000.hex.txt)

closed 2020-11-10 by SukkoPera