Optiboot/optiboot #301: Optiboot grows bigger with (Arduino IDE) avr-gcc 7.3.0

WestfW, 2020-11-10:
Compiled with the version of avr-gcc (7.3.0) distributed with recent Arduino IDEs (1.8.13), optiboot grows 14 bytes.
The normal versions still fit in 512 bytes, but attempts to build a non-blinking, EEPROM-support version (documented as possible) fail...

The area of code growth is not obvious :-(

See https://github.com/Optiboot/optiboot/issues/300

```
 make GCCROOT=/Applications/Arduino-1.8.13.app/Contents/Java/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/ atmega328
avr-gcc (GCC) 7.3.0

/Applications/Arduino-1.8.13.app/Contents/Java/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3              -c -o optiboot.o optiboot.c

/Applications/Arduino-1.8.13.app/Contents/Java/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3            -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o 
/Applications/Arduino-1.8.13.app/Contents/Java/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-size optiboot_atmega328.elf
   text    data     bss     dec     hex filename
    496       0       0     496     1f0 optiboot_atmega328.elf
```
```
 make atmega328
avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.1_495) 5.4.0

avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3              -c -o optiboot.o optiboot.c

avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3            -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o 
avr-size optiboot_atmega328.elf
   text    data     bss     dec     hex filename
    482       0       0     482     1e2 optiboot_atmega328.elf
```

WestfW, 2020-11-10:
Huh.  optiboot_x SHRINKS by two bytes.

WestfW, 2020-11-10:
avr-gcc 5.4:
```
00007fc0 00000020 t do_spm
00007fb2 0000000e t getNch
00007f7e 00000018 T getch
00007e04 0000016c T main
00007ffe 00000002 R optiboot_version
00007e00 00000004 T pre_main
00007f70 0000000e T putch
00007fa2 00000010 T verifySpace
00007f96 0000000c T watchdogConfig
```

avr-gcc 7.3.0:
```
00007fce 00000020 t do_spm
00007fc0 0000000e t getNch
00007f8c 00000018 T getch
00007e04 0000017a T main
00007ffe 00000002 R optiboot_version
00007e00 00000004 T pre_main
00007f7e 0000000e T putch
00007fb0 00000010 T verifySpace
00007fa4 0000000c T watchdogConfig
```

motolav, 2021-08-23:
I ran "make atmega328p" with GCC 11.1.0 and it shrunk 2 bytes compared to your 7.3.0
GCC 12 will have the mode_cc changes to the AVR target, which is already in master

mcuee, 2022-12-19:
Please refer to here as well.
* https://github.com/Optiboot/optiboot/issues/355#issuecomment-1345109821

https://user-images.githubusercontent.com/1294740/206854595-e45ac46e-ec61-483d-aa50-96d3c8cc8166.png

mirh, 2025-05-26:
There were a lot of screwups fixed in the last year
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=90706
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=112268
(and even for [the one](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=114243) I still found open, there's a workaround flag)