Optiboot/optiboot #303: 8MHz/3.3V Cannot upload over UART with optiboot but can with Arduino bootloader

brolly759, 2021-01-13:
Running into a strange issue where the optiboot bootloader refuses to upload code over USB/UART but works find with default Arduino Pro Mini 3.3v/8MHz board settings and bootloader.

Hardware: Custom PCB (ATMEGA328P-MU-ND) and Programmer (cp2102N)
ALT Hardware: 2 Arduino Nano Pro Mini's + FTDI 
Arduino IDE 1.8.12
avrdude: Version 6.3-20190619

I am using the board manager with v8.0 optiboot.
I select optiboot on 32-pin MCUs
I select 8MHz(int)
I use Arduino as ISP to upload bootloader to other Arduino Nano's and custom PCB.

- Uploading to Arduino Nano Pro Mini, everything works. I can upload firmware over UART/USB using an FTDI chip with latest optiboot settings.
- Uploading to Custom PCB with custom programmer, cannot upload new FW over USB/UART. 
- I used AVRDUDE to push a FW+Bootloader with serial.print() and the program is running/going into the void setup/loop functions so its not locked. 
- I have a 10K pullup from RESET pin to VCC. 
- Thinking it was a timing issue, I changed the DTR  capacitor to 1uF, .1uF, .01uF, no difference. 

At this point I would figure that its a hardware issue but I load the original bootloader (non optiboot) from Arduino IDE and everything works fine and I can upload over UART/USB. 

So what am I missing? Is there something in the bootloader that is checking for pin states? Is there still a timing issue somewhere? Why would the non optiboot bootloader work? I even thought it was an issue with the LED start up routine so I built my own bootloader with this command: make UART=1 AVR_FREQ=8000000L BAUD_RATE=19200 LED_START_FLASHES=0 atmega328. I also tested MiniCore and same issues.

Any help would be appreciated! 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM22
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
         Hardware Version: 4744608
         Firmware Version: 8.4611299
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
         Vtarget         : 420030.5 V
         Varef           : 0.3 V
         Oscillator      : 921.600 kHz
         SCK period      : 10844687.6 us

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

brolly759, 2021-01-13:
Update. I changed the USB programmer from the CP2102N to the FTDI, added a 100nF Cap to DTR and I can program over USB now. So there is something wrong with the CP2102N wiring, driver or chipset itself?

cimba007, 2021-04-02:
Have the same problem here. Still investigating.

WestfW, 2021-04-04:
Exactly which .hex file or build options are you using?
57600 is one of the bit rates that isn't exact at 8Mz (about 2.1% off) and it's possible that some USB/Serial converters might be off in the OTHER direction, causing problems.

closed 2021-04-21 by WestfW