Optiboot/optiboot #313: Cosmetic reformatting.

WestfW, 2021-05-02:
The formatting (indentation, use of tabs vs spaces, etc) of optiboot.c has become very inconsistent over the years.
Both the SpenceKonde and MCUDude branches are in better shape; the "master" repository ought to get its act together:

1. 2-space indent default
2. no tabs
3. fix trailing whitespace
4. fix spelling mistakes
5. do something to make auto-indent work properly in spite of the conditional compilation

Do this separately from any actual code changes!

per1234, 2021-05-02:
In the case of the SpenceKonde repos ([example](https://github.com/SpenceKonde/megaTinyCore/blob/master/.github/workflows/check-code-formatting.yml)), I used the Artistic Style code formatter with [a configuration](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ci/arduino-code-style.conf) that is based on the one used for formatting the IDE's built-in example sketches:
https://github.com/arduino/arduino-examples/blob/main/examples_formatter.conf
which is in turn based on the configuration used by the IDE's Auto Format:
https://github.com/arduino/Arduino/blob/master/build/shared/lib/formatter.conf

However, the Arduino IDE 2.x uses ClangFormat instead of Artistic Style for Auto Format. You can see the configuration here:
https://github.com/arduino/arduino-language-server/blob/64a0e42a7a3d713629d3066d7555fb32cf19c3ec/handler/handler.go#L1776-L1919
This configuration is intended to be as close as possible to the previous Artistic Style configuration, but ClangFormat doesn't allow so much room to disable formatting options like Artistic Style does, so in some cases the new configuration mandates a specific style where the old one allowed the user to use any style they wanted.

In the case of the MCUdude repositories, I think we only used some more general formatting checks:
https://github.com/MCUdude/MegaCore/blob/v2.1.2/.travis.yml#L24-L39
Even in the SpenceKonde repositories where the code formatting check was in use, I did those general formatting checks as well:
https://github.com/SpenceKonde/megaTinyCore/blob/master/.github/workflows/general-formatting-checks.yml
The reason being that the code formatter only checks the formatting of code files, whereas I want to apply these general formatting requirements to all files

I'm shifting from that approach to using `.editorconfig` with https://github.com/editorconfig-checker/editorconfig-checker, as documented here:
https://github.com/per1234/.github/blob/development/workflow-templates/check-general-formatting.md
The things I miss from the new approach are:
- Check for use of tabs for alignment (it only checks indentation style)
- File encoding check.

per1234, 2021-05-02:
I shared some information about the tools for bringing the repository into compliance here:
https://github.com/SpenceKonde/megaTinyCore/issues/242#issuecomment-706634475

WestfW, 2021-05-15:
https://github.com/Optiboot/optiboot/commit/564372673a5167b3e30da1f3e4e29916321a9b20#diff-ef40075c08d90ece6c084ec57b2bf83cdbfea2afa3ef22b0b66354470f6ca79b
(I haven't checked against the style checkers; I really only care about internal consistency and EMACS :-) )

closed 2021-05-15 by WestfW