Optiboot/optiboot #317: Using Atmel to Build bootloader

BeminGhobreal, 2021-06-16:
Hi All, I'm working with optiboot and trying to build it via atmel studio. It uses the make.atml 
but it fails to build and it gives the following output 

`------ Rebuild All started: Project: xplained328pb_6_2, Configuration: Release AVR ------
Build started.
Project "xplained328pb_6_2.cproj" (Clean target(s)):
Target "Clean" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Gitrepo\Bootloader Example\optiboot\optiboot\AtmelStudio\xplained328pb_6_2.cproj" (entry point):
	Using "RunCompilerTask" task from assembly "C:\Program Files (x86)\Atmel\Studio\7.0\Extensions\Application\AvrGCC.dll".
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\Gitrepo\Bootloader Example\optiboot\optiboot\AtmelStudio\..\bootloaders\optiboot" -f "Makefile" clean 
		make: Entering directory 'C:/Gitrepo/Bootloader Example/optiboot/optiboot/bootloaders/optiboot'
		rm -rf *.o *.elf *.lst *.map *.sym *.lss *.eep *.srec *.bin *.hex *.tmp.sh
C:\Gitrepo\Bootloader Example\optiboot\optiboot\bootloaders\optiboot\Makefile(672,1): error: recipe for target 'clean' failed
		make: *** [clean] Error -1073741502
		make: Leaving directory 'C:/Gitrepo/Bootloader Example/optiboot/optiboot/bootloaders/optiboot'
	Done executing task "RunCompilerTask" -- FAILED.
Done building target "Clean" in project "xplained328pb_6_2.cproj" -- FAILED.
Done building project "xplained328pb_6_2.cproj" -- FAILED.

Build FAILED.
------ Rebuild All started: Project: xplained328pb_6_2, Configuration: Release AVR ------
Build started.
Project "xplained328pb_6_2.cproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Gitrepo\Bootloader Example\optiboot\optiboot\AtmelStudio\xplained328pb_6_2.cproj" (target "Build" depends on it):
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\Gitrepo\Bootloader Example\optiboot\optiboot\AtmelStudio\..\bootloaders\optiboot" -f "Makefile" xplained328pb 
		make: Entering directory 'C:/Gitrepo/Bootloader Example/optiboot/optiboot/bootloaders/optiboot'
		"C:/Program Files (x86)/Atmel/Studio/7.0/shellUtils/make" atmega328 AVR_FREQ=16000000L BAUD_RATE=57600
C:\Gitrepo\Bootloader Example\optiboot\optiboot\bootloaders\optiboot\Makefile.atmel(41,1): error: recipe for target 'xplained328pb' failed
		make: *** [xplained328pb] Error -1073741502
		make: Leaving directory 'C:/Gitrepo/Bootloader Example/optiboot/optiboot/bootloaders/optiboot'
	Done executing task "RunCompilerTask" -- FAILED.
Done building target "CoreBuild" in project "xplained328pb_6_2.cproj" -- FAILED.
Done building project "xplained328pb_6_2.cproj" -- FAILED.

Build FAILED.
========== Rebuild All: 0 succeeded, 1 failed, 0 skipped ==========`

I know the make file works, If I run make from the command line it works, as following :

`"C:/Program Files (x86)/Atmel/Studio/7.0/shellUtils/make" atmega328 AVR_FREQ=16000000L BAUD_RATE=57600`

So it must has to do with ATMEL Studio 
I tried adding  the below two lines but it didn't work.
`xplained328pb: CFLAGS += -B "C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\ATmega_DFP\1.2.209\gcc\dev\atmega328pb"
xplained328pb: CFLAGS += -I"C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\ATmega_DFP\1.2.209\include"`

Any Suggestions ?

WestfW, 2021-06-17:
It's odd that it's failing on the "make clean" part as well, and with such odd error codes.
It works fine here (this on a VM that shouldn't have much customization...)

```txt
------ Rebuild All started: Project: xplained328p_6_2, Configuration: Debug AVR ------
Build started.
Project "xplained328p_6_2.cproj" (Clean target(s)):
Target "Clean" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Users\billw\Downloads\optiboot2021\optiboot-master\optiboot\AtmelStudio\xplained328p_6_2.cproj" (entry point):
	Using "RunCompilerTask" task from assembly "C:\Program Files (x86)\Atmel\Studio\7.0\Extensions\Application\AvrGCC.dll".
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\Users\billw\Downloads\optiboot2021\optiboot-master\optiboot\AtmelStudio\..\bootloaders\optiboot" -f "Makefile" clean 
		make: Entering directory 'C:/Users/billw/Downloads/optiboot2021/optiboot-master/optiboot/bootloaders/optiboot'
		rm -rf *.o *.elf *.lst *.map *.sym *.lss *.eep *.srec *.bin *.hex *.tmp.sh
		make: Leaving directory 'C:/Users/billw/Downloads/optiboot2021/optiboot-master/optiboot/bootloaders/optiboot'
	Done executing task "RunCompilerTask".
Done building target "Clean" in project "xplained328p_6_2.cproj".
Done building project "xplained328p_6_2.cproj".

Build succeeded.
------ Rebuild All started: Project: xplained328p_6_2, Configuration: Debug AVR ------
Build started.
Project "xplained328p_6_2.cproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Users\billw\Downloads\optiboot2021\optiboot-master\optiboot\AtmelStudio\xplained328p_6_2.cproj" (target "Build" depends on it):
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\Users\billw\Downloads\optiboot2021\optiboot-master\optiboot\AtmelStudio\..\bootloaders\optiboot" -f "Makefile" xplained328p 
		make: Entering directory 'C:/Users/billw/Downloads/optiboot2021/optiboot-master/optiboot/bootloaders/optiboot'
		"C:/Program Files (x86)/Atmel/Studio/7.0/shellUtils/make" atmega328 AVR_FREQ=16000000L BAUD_RATE=57600
		make[1]: Entering directory 'C:/Users/billw/Downloads/optiboot2021/optiboot-master/optiboot/bootloaders/optiboot'
		avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1778) 5.4.0
		Copyright (C) 2015 Free Software Foundation, Inc.
		This is free software; see the source for copying conditions.  There is NO
		warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
		BAUD RATE CHECK: Desired: 57600, Real: 57142, UBRRL = 34, Difference=0.7%
		avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3                 -c -o optiboot.o optiboot.c
		avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3               -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o 
		avr-size optiboot_atmega328.elf
		   text	   data	    bss	    dec	    hex	filename
		    482	      0	      0	    482	    1e2	optiboot_atmega328.elf
		avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
		avr-objdump -h -S optiboot_atmega328.elf > optiboot_atmega328.lst
		rm optiboot.o
		make[1]: Leaving directory 'C:/Users/billw/Downloads/optiboot2021/optiboot-master/optiboot/bootloaders/optiboot'
		mv optiboot_atmega328.hex optiboot_xplained328p.hex
		mv optiboot_atmega328.lst optiboot_xplained328p.lst
		make: Leaving directory 'C:/Users/billw/Downloads/optiboot2021/optiboot-master/optiboot/bootloaders/optiboot'
	Done executing task "RunCompilerTask".
	Using "RunOutputFileVerifyTask" task from assembly "C:\Program Files (x86)\Atmel\Studio\7.0\Extensions\Application\AvrGCC.dll".
	Task "RunOutputFileVerifyTask"
		
		Display Output File Size Skipped due to : Output File not found
	Done executing task "RunOutputFileVerifyTask".
Done building target "CoreBuild" in project "xplained328p_6_2.cproj".
Target "PostBuildEvent" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Avr.common.targets" from project "C:\Users\billw\Downloads\optiboot2021\optiboot-master\optiboot\AtmelStudio\xplained328p_6_2.cproj" (target "Build" depends on it):
	Task "Exec"
		copy optiboot_xplained328p.*  ..\..\AtmelStudio

		optiboot_xplained328p.hex
		optiboot_xplained328p.lst
		        2 file(s) copied.
	Done executing task "Exec".
Done building target "PostBuildEvent" in project "xplained328p_6_2.cproj".
Target "Build" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Avr.common.targets" from project "C:\Users\billw\Downloads\optiboot2021\optiboot-master\optiboot\AtmelStudio\xplained328p_6_2.cproj" (entry point):
Done building target "Build" in project "xplained328p_6_2.cproj".
Done building project "xplained328p_6_2.cproj".

Build succeeded.
========== Rebuild All: 1 succeeded, 0 failed, 0 skipped ==========
```

BeminGhobreal, 2021-06-24:
What version of ATMEL are you using? When opening the .sln file Atmel requested upgrading the sln file and that could have caused that issue.
[image]

BeminGhobreal, 2021-10-08:
> I've added an AS7 "Solution" and several "Projects" that should help.
> c9bbfb2
> #diff-d00465a0b6fb5a959e455fec9bbf7ab1553c0720e2a736d94b4a1d5a7156082b

Hi Bill,

Thanks I resolved the issue