Optiboot/optiboot #319: PC Tool to communicate with optiboot to flash the Application

BeminGhobreal, 2021-07-09:
HI all, I'm working on a  custom PC tool to upload the flash. I'm using the atmega328pb and the optiboot bootloader, I'm trying to use the stk500 command but the bootloader not responding. The issue is in the my pc application because the bootloader working with averdude.
I couldn't find any documentation or command protocol regarding communicating with optiboot from the PC  ?
Any suggestions ?

WestfW, 2021-07-09:
Is your PC-side code using STK500v1 or STK500v2?  They're substantially different.
See https://github.com/Optiboot/optiboot/wiki/HowOptibootWorks

BeminGhobreal, 2021-07-15:
Using STKV500v1

BeminGhobreal, 2021-07-15:
I can't find anything has to do with the crc ? how is the crc calculated

BeminGhobreal, 2021-07-23:
How to program flash memory using optiboot ?

WestfW, 2021-07-24:
There is no actual CRC.  the "20" is more of a "check character", despite the symbol name.

What part of the previous references are you stuck on?  In between the wiki page and the Atmel App note, it should be pretty clear, and I don't know of anything else.

BeminGhobreal, 2021-07-24:
I resolved all the above, but now My issue is when I program the flash memory I don't get the /14/10 return , which I'M SUPPOSE TO GET ?

closed 2022-03-11 by WestfW