Optiboot/optiboot #321: The bootloader seams to doesn't work at all with my version of avrdude

Virtual-Java, 2021-07-17:
Compiling (avrgcc-5.4) and flashing (1.) the original optibootloader for/to atmega328 was no problem, but uploading e.g. Blink.ino.hex to a device that contains optiboot (2.) doesn't work (with my version of avrdude). However uploading Blink.ino.hex (avrdude via cmdline) to a mcu with the Arduino IDE's optiboot-version on it is successful!! My Fuses for optiboot (m328p) are shown in 3.).
1.) 
`~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/etc/avrdude.conf -c usbasp -p atmega328p -P usb -e -u -U efuse:w:0xFE:m -U hfuse:w:0xDE:m -U lfuse:w:0xFF:m -U flash:w:optiboot_atmega328.hex -U lock:w:0x2f:m`

2.)
`/home/jonathan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/jonathan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega328p -c stk500v1 -P /dev/ttyACM0 -b9600 -D -Uflash:w:/tmp/arduino_build_259003/Blink.ino.hex:i`
...         _Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.03s
avrdude: Device signature = 0x000000 (retrying)
Reading | ################################################## | 100% 0.04s
avrdude: Device signature = 0x000000 (retrying)
Reading | ################################################## | 100% 0.04s
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.
avrdude done.  Thank you._

3.)
`~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/etc/avrdude.conf -c usbasp -p atmega328p -P usb -b 115200 -U efuse:r:-:i -U hfuse:r:-:i -U lfuse:r:-:i -U lock:r:-:i -v`
_avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DE
avrdude: safemode: efuse reads as FE
avrdude: safemode: Fuses OK (E:FE, H:DE, L:FF)_

WestfW, 2021-07-17:
>> avrdude  -v -patmega328p **-c stk500v1**

You should need "-c arduino" to talk to optiboot.   Avrdude's stk500v1 support was changed a long time ago to use "raw spi commands" for a lot of the functions (like "read signature"), and optiboot doesn't support those...

Virtual-Java, 2021-07-18:
BURNING Optiboot (precompiled optiboot_atmega328p) bootloader via avrdude on the cmdline:
`$ # ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/etc/avrdude.conf -c usbasp -P usb -p atmega328p -e -u -U flash:w:optiboot_atmega328.hex:i -U efuse:w:0xFE:m -U hfuse:w:0xDE:m -U lfuse:w:0xFF:m -U lock:w:0x2f:m`

_... avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: reading input file "optiboot_atmega328.hex"
avrdude: writing flash (32768 bytes): ..._

UPLOADING Blink.ino to Arduino Uno Board with Optiboot precompiled bootloader (optiboot_atmega328.hex) using avrdude on the cmdline:
`$ # /home/jonathan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/jonathan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega328p -c arduino -P /dev/ttyACM1 -b115200 -D -Uflash:w:/tmp/arduino_build_534594/Blink.ino.hex:i`
_avrdude: Version 6.3-20190619 ...
avrdude: verifying ...
avrdude: 3894 bytes of flash verified
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)
avrdude done.  Thank you._

Burning and uploading with the scheme above does work reproducible!!!
Thank you very much for the solution and your very fast response.

Virtual-Java, 2021-07-18:
By the way my self compiled version (with gcc-11.1.0) of optiboot works too.
Now I hope my version of Optiboot for AVR-USB-MCUs like the Atmega32u4 will work too. :)
https://github.com/Virtual-Java/optiboot-usb-mcus

WestfW, 2021-07-18:
Oh!  I'd heard that newer avr-gccs were generally producing larger code; are you confirming that it's still under 512 bytes even in gcc11?  That's useful data!

Virtual-Java, 2021-07-18:
Recently when I was working on a modified version of HoodLoader I had issues dealing the size of bootloader compiled with gcc10.1. https://github.com/NicoHood/HoodLoader2/issues/93
For this reason I am considerably interested in finding the reason why newer versions of gcc produce pretty much larger code.
Especially newer and more special devices with lots of memory or peripherals like the Atmega32u4 cause the boot size to diverge increasingly.
With avr-gcc5.4 HoodLoader2.0.5 is 59 Bytes smaller compared to avr-gcc10.1 (4050 bytes vs 4109 bytes in size).
consequently the bootloader didn't fit into the 4096 bytes (4kB) maximum size for boot section.
https://github.com/Virtual-Java/HoodLoader2/compare/mint-gcc5...Virtual-Java:virtual-5kB-bootsize-test
My temporary work around was using the older version of gcc running on Linux Mint, but my favorite distribution Arch comes with gcc10.1. Unfortunately gcc version 5.4 is not available in AUR so I tried to convert a debian package to pacman format, but I wasn't successful.
Maybe there are some additional compiler options (excluding Os and relax) for backwards compatibility we can turn on to decrease compile size?

Virtual-Java, 2021-07-19:
Sorry, I had forgotten to answer your question whether the code produced with newer avr-gcc fits into 512 bytes (256 words) sized bootsection. The answer is yes, optiboot compiled for atmega328p fits into 512 bytes bootsection with each of them, gcc5.4.0 or gcc11.1.0, which results in a code size of 492 respectively 494 bytes.

Besides the version of avr-gcc I used was 11.1.0 and NOT 10.1 as I wrote in my last post.

To make it easy to compare the output between these two versions of avr-gcc, I compiled Optiboot for all targets listed in the makeall scripts one time with gcc5.4.0 and another time with gcc11.1.0.

The output of the compilation process is resided in optiboot/bootloaders/optiboot/compilelog.<platform>.txt.
Hopefully that will help.

Virtual-Java, 2021-07-19:
Comparison between gcc5 and gcc10: https://github.com/Virtual-Java/optiboot-usb-mcus/compare/mint-gcc5...Virtual-Java:arch-gcc10

closed 2021-07-30 by WestfW