Optiboot/optiboot #325: Upload to AtMega4809 on Arduino Uno WiFi2 fails

Virtual-Java, 2021-09-07:
After downloading the extension packages from microchip, compiling the hex file worked for me. Unfortunately uploading to the Arduino Uno WiFi fails reproducibly (Programmer is not responding).
Previous the IO-MCU on the board was flashed with Optiboot via the onboard UPDI Programmer running on the AtMega32u4 USB-MCU.
For my tests the AtMega4809  is connected via the UART pins broken out to pins D0 (RX) and D1 (TX) to another Arduino Uno (without a processor in the socket) being used as a USB to UART converter.
I have checked my connections several times, so the hardware should do its job.
But maybe my settings for the fuses are wrong?

**Compiling Optiboot for "Mega0 AVR boards":**
**1.) Download** Atmel.ATmega_DFP.<newest.version>.atpack.zip from Microchips website (e.g. `Atmel.ATmega_DFP.1.7.374.atpack.zip` from http://packs.download.atmel.com)
**2.) unzip Atmel.ATmega_DFP.<newest.version>.atpack.zip** to any directory and remember the path to it (e.g. `\~/Downloads/Atmel.ATmega_DFP.<newest.version>.atpack_Files`)
**3.) export the path and adapt the path in all makefiles** when using make scripts like makeall.mega0.sh (e.g. `export PACKS=\~/Downloads/Atmel.ATmega_DFP.1.7.374.atpack_FILES`)
**4.) compile for the new Atmega devices either manually or by running makefiles** (e.g. `make -f Makefile.mega0 UARTTX=C4 LED=D6 atmega4809` or `. makeall.mega0.sh`
**5.) NO error messages** caused by missing packs should appear during compilation!!!_

  _**Bootloader size with skeleton App**
  text	   data	    bss	    dec	    hex	filename
  04             9              0	      513	     201	optiboot_atmega4809.elf
    avr-size optiboot_atmega4809.hex
  **Bare Bootloader size**
  text	   data	    bss	    dec	    hex	filename
     0	     466	      0	      466	    1d2	optiboot_atmega4809.hex_

**6.) Rename the created files:**
for most linux distros like Arch:`'rename optiboot_atmega4809 optiboot_atmega4809_UnoWiFi2 optiboot_atmega4809.*`
for Ubuntu derivates like Mint: `rename 's/optiboot_atmega4809/optiboot_atmega4809_UnoWiFi2/;' optiboot_atmega4809.*`
**7.) create custom hexfile directory:** `mkdir ../../hexfiles/mega0_custom`
**8.) move UnoWiFi2 files to hexfiles directory:** `mv optiboot_atmega4809_UnoWiFi2.* ../../hexfiles/mega0_custom/`
**9.) test image by uploading it to Arduino Uno WiFi Rev2:**

10.)**Backup the fuses and flash memory (dump flash):**
Optiboot/compile_optiboot_gcc5_4/original-github/optiboot/optiboot/hexfiles/mega0_custom$ ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -p atmega4809 -c xplainedmini_updi -P usb -D -U fuse0:r:fuse0:b -U fuse1:r:fuse1:b -U fuse2:r:fuse2:b -U fuse5:r:fuse5:b -U fuse6:r:fuse6:b -U fuse7:r:fuse7:b -U fuse8:r:fuse8:b -U flash:r:optiboot_atmega4809_UnoWiFi2_dump.hex:i

11.) **My commands to BURN Optibootloader:**
AVRdude burn custom Optiboot bootloader: \~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C \~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega4809 -cxplainedmini_updi -Pusb -Ufuse0:w:0x00:m -Ufuse1:w:0x48:m -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse6:w:0x07:m -Ufuse7:w:0x00:m -Ufuse8:w:0x02:m -Ufuse9:w:0xC5:m -Uflash:w:optiboot_atmega4809_UnoWiFi2.hex:i

12.)**RESTORE original bootloader dumped from flash previously (works):**
Arduino IDE burn original mEDBG bootloader: \~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C \~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega4809 -cxplainedmini_updi -Pusb -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x02:m -Uflash:w:\~/.arduino15/packages/arduino/hardware/megaavr/1.8.7/bootloaders/atmega4809_uart_bl.hex:i

Q1:) FUSE VALUES: are they correct???
-Ufuse0:w:0x00:m    Watchdog timer window and period mode are disabled
-Ufuse1:w:0x48:m    BODLEVEL2 2.6V; SampleFreq 1kHz; Sampled in Active and Idle Mode only
-Ufuse2:w:0x01:m    Oscillator calibration at runtime is possible; internal oscillator running at 16MHz
-Ufuse3DONTWRITE    RESERVED
-Ufuse4DONTWRITE    RESERVED
-Ufuse5:w:0xC9:m    no CRC Source; RESET pin; keep EEPROM during chip erase
-Ufuse6:w:0x07:m    64 ms startup time after power on before code execution
-Ufuse7:w:0x00:m    append  is at flashend otherwise size of bootsection + appsection (rest unused)
-Ufuse8:w:0x02:m    bootend is 512 Bytes (256 words) from flashstart
-Ufuse9:w:0xC5:m    device is open, reading all memories is possible
Q2:) ERROR: "fuse9" memory type not defined for part "ATmega4809" WHY???

 Q3:) Why is my arduino USB-MCUs Serial number FFF…
         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         Serial number   : FFFFFFFFFFFFFFFFFFFF
         Vtarget         : 5.00 V

Since I spent a lot of time to research the information how to compile Optibootloader for the AtMega4809, I decided to describe my procedure very detailed to simplify the steps for other people.

Because I am not sure how many mistakes I made, I will definitely correct them afterwards. 
Please tell me if I should change my original post as far as possible or post an improved short guide.
Maybe I can spend some time in writing a file for make atmega4809-isp, when the fuses are clear.ed:)

WestfW, 2021-10-04:
I don't understand...
> uploading to the Arduino Uno WiFi fails reproducibly

Uploading, how?

> RESTORE original bootloader dumped from flash previously

Isn't the original bootloader only for OTA updates, via the wifi module (which is on a different UART than pins0/1)
It also uses a different protocol than optiboot, based on [AN2634](https://ww1.microchip.com/downloads/en/Appnotes/AN2634-Bootloader-for-tinyAVR-and-megaAVR-00002634C.pdf)

closed 2022-03-11 by WestfW

mcuee, 2022-12-10:
Maybe this is related. 
https://github.com/arduino/ArduinoCore-megaavr/issues/119