Optiboot/optiboot #334: Cannot upload reliably using Bluetooth and an HC-05

mgurzixo, 2021-11-30:
When connecting an Arduino pro mini to an HC-05, I was not able to upload a sketch reliably OTA.
The long story is [there](https://github.com/mgurzixo/flashota) .

My solution was to patch the function `verifySpace()` inside Optiboot like this:

```
void verifySpace() {
  if (getch() != CRC_EOP) {
#ifdef CLASSIC_VERIFY_SPACE
    watchdogConfig(WATCHDOG_16MS);    // shorten WD timeout
    while (1)            // and busy-loop so that WD causes
      ;              //  a reset and app start.
#else
    putch(STK_NOSYNC);
#endif
  }
  else {
    putch(STK_INSYNC);
  }
}
```

So that, if there is noise on the line, Optiboot sends back STK_NOSYNC instead of resetting, allowing AVRDUDE to resynchronize nicely.
Doing that allows now for a very reliable OTA sketch upload.

Any comments?

Michel

WestfW, 2021-12-01:
The point of aborting on "illegal" characters is to prevent looping inside the bootloader in the case where it is talking to some traffic generator OTHER than uploading software...

Is OTA really that noisy?  With some specific character, or just random noise?

mgurzixo, 2021-12-02:
Thanks WestfW for the answer and writing this SUPERB piece of code.

This makes sense....

Unfortunately I have no idea of why I am obliged to do this patch, and I don't have a logic analyser. I just know that it works perfectly reliably ;)

I am not positive about this, but I think that the watchdog is still active, so if a valid command is not received within 1second, it reboots. If that is the case, the problem is already solved.

Anyway, if you don't accept this patch, I suggest that at least you add it as an option turned off by default with a link to this thread, so that people wanting to flash Arduinos OTA through Bluetooth ( if you google that, you will get tens of people who tried and failed miserably) get a pointer and know that it is possible :)

mgurzixo, 2021-12-04:
Just to be sure, I tried to comment out `putch(STK_NOSYNC);`, so that Optiboot returns nothing in case of wrong command, but this does not work. 

In that case, I get systematically the following output:
```
$ /home/XXX/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C/home/XXX/Arduino/hardware/MgCore/avr/avrdude.conf -v -patmega328p -carduino -P/dev/rfcomm0 -b115200 -D -Uflash:w:/home/XXX/Arduino/flashota/.build/flashota.ino.hex:i
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/XXX/Arduino/hardware/MgCore/avr/avrdude.conf"
         User configuration file is "/home/XXX/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/rfcomm0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x08

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 4238814
         Firmware Version: 18231472.0
         Topcard         : Unknown
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x08
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.
```
This proves that there is a synchronization problem somewhere.
My conclusion is that the ONLY solution for reliable OTA uploading using vanilla `AVRDUDE` and HC-05 is to use this patch.