Optiboot/optiboot #336: Issue with Atmega8

dakhnod, 2021-12-27:
When flashing my Atmega8 using the command line
`make atmega8_isp AVR_FREQ=8000000L BAUD_RATE=57600 LFUSE=E4 HFUSE=DC LED_START_FLASHES=3 LED=B0 BAUD_RATE=9600`
the LED blinks, thus the bootloader is started.
After a power cycle the LED doesn't blink, so the MCU doesn't start for some reason.

When setting the boot flash section to 256 words, the boot section start is 0x0F00.
When using objdump with the hex (attached), the address in the hex seem to start at 0x1E00:
` 1e00 01c0d6c0 11248fe5 94e09ebf 8dbf84b7  .....$..........`
Shouldn't the bootloader code be written to 0x0F00?

dakhnod, 2021-12-28:
My bad, I just had to run make with BOOT_ON_POR=1, that fixed it.

Still, could someone please explain to me why the starting address in the hex file does not math the flsah address of the bootloader in the MCU?

WestfW, 2021-12-28:
Atmel documents code addresses in WORDS, but all the gnu tools use/specify the address in bytes.
word 0xF00 ==  byte 0x1E00

closed 2021-12-29 by WestfW