Optiboot/optiboot #338: Microchip Studio and Atmega328pb

SandeepKel, 2022-02-16:
Trying to build the project for Atmega328pb with custom bootloader.

The build works correctly with built-in makefile, the moment i redirect to custom  make file, it fails to build attached error and make file for your reference
[image]
[Makefile.txt](https://github.com/Optiboot/optiboot/files/8078494/Makefile.txt)
.

WestfW, 2022-02-19:
I think you want CFLAGS added to your link command as well; it might need stuff that is part of the "pack."  (although normally the bootloader build goes to avoid linking in the standard startup code via -nostartfiles, so this might not be your only problem.)

SandeepKel, 2022-02-21:
Hey WestFW, thank you for your comments, the build is successful after adding the CFLAGS while linking.

closed 2022-02-21 by SandeepKel