Optiboot/optiboot #347: Support Tiny-2 series MCUs

tpimh, 2022-06-10:
As I understand from this line:
https://github.com/Optiboot/optiboot/blob/81bdd404ff70b76fed716a10bbde90149d98e647/optiboot/bootloaders/optiboot/Makefile.mega0#L1
Only Mega-0, Tiny-0, and Tiny-1 families are supported. What modifications are actually needed to compile Optiboot_X for the following list of MCUs: `ATtiny{4,8,16,32}2{4,6,7}`?

Tiny-2 family list expanded:
- ATtiny424
- ATtiny426
- ATtiny427
- ATtiny824
- ATtiny826
- ATtiny827
- ATtiny1624
- ATtiny1626
- ATtiny1627
- ATtiny3224
- ATtiny3226
- ATtiny3227

WestfW, 2022-06-11:
That appears to be true.
It looks like "DrAzzy" had added those to his copy of optiboot_x:
https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/bootloaders/optiboot_x

WestfW, 2022-07-19:
Fixed: https://github.com/Optiboot/optiboot/commit/45d4c1a5207bacff5de18d8ab5d2a734435c9644

closed 2022-07-19 by WestfW