Optiboot/optiboot #349: make ATTINY85 with RS485 retun error

expertup, 2022-06-17:
While I run command:
`make attiny85 RS485=B2 LED_START_FLASHES=0 TIMEOUT=8 BAUD_RATE=19200 NO_START_APP_ON_POR=1 START_APP_ON_EXTR=1`
return this error:
```
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=attiny85 -DF_CPU=8000000L  -DBAUD_RATE=19200 '-DLED_START_FLASHES=0'           -DWDTTIME=8 -DNO_START_APP_ON_POR=1 -DSTART_APP_ON_EXTR=1 -DRS485=B2 '-DVIRTUAL_BOOT_PARTITION' '-DSOFT_UART' -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_attiny85.elf optiboot.o
c:/users/dell/programs/avr8-gnu-toolchain/bin/../lib/gcc/avr/12.1.0/../../../../avr/bin/ld.exe: optiboot_attiny85.elf section `.text' will not fit in region `text'
c:/users/dell/programs/avr8-gnu-toolchain/bin/../lib/gcc/avr/12.1.0/../../../../avr/bin/ld.exe: section .version LMA [0000000000001ffe,0000000000001fff] overlaps section .text LMA [0000000000001dc0,0000000000002005]
c:/users/dell/programs/avr8-gnu-toolchain/bin/../lib/gcc/avr/12.1.0/../../../../avr/bin/ld.exe: region `text' overflowed by 0 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [optiboot_attiny85.elf] B▒▒d 1
rm optiboot.o
```

But if I run without `RS485=B2` pogram is compling.

I tried to add `attiny85: LDSECTIONS  = -Wl,--section-start=.text=0x1c00` in `Makefile.tiny` but problem is after connect attiny85 to power, the microcontroller runs the main program without the bootloader. In the project I need to disable the reset pin and use it as an additional pin. So the bootloader must start after attiny85 is powered up.

WestfW, 2022-06-18:
Try an older version of gcc.  The latest Microchip version is avr-gcc 5.4.something, and Arduino is support 7.3.something.
12.1.0 is WAY "out there", and I'm not inclined to spend time analyzing/debugging/fixing optiboot to work with it.  It's disappointing that it apparently causes code size to grow so much - it's 568 bytes with v5.4 (Hmm.  Ok, it's not THAT much bigger - the allocated size turns out to be 576 bytes...)

Note that the mechanism for specifiying bootloader size and start address has changed recently, so that it's now in the C code instead of being controlled by section-start linker commands.  See the optiboot.c code near "sectionOpts()" -just bump up VBSIZE...

mcuee, 2022-12-10:
FYI, latest Atmel toolchain is now based on avr-gcc 7.3.
https://www.microchip.com/en-us/tools-resources/develop/microchip-studio/gcc-compilers

The older avr-gcc 5.4 based version is still available though.

mcuee, 2022-12-10:
Interestingly urboot only recommends avr-gcc 4.8.1 and avr-gcc 5.4 now.

But I think later avr-gcc 7.3 based toolchain will be used for UPDI parts support.

mcuee, 2022-12-10:
As of now, Arduino shiped toolchain works with optiboot/optiboot_x.

For optiboot_dx, you may want to use the toolchain from DxCore. It also works with optiboot/optiboot_dx.
https://github.com/avrdudes/avrdude/issues/1200#issuecomment-1328390232

You can also use Atmel toolchain as the base and manually copy the necessary files from the DFPs from Atmel Studio but that is a hassle.

mcuee, 2022-12-10:
FYI, MSYS2 comes with avr-gcc 8.5 and it seems to work as well.
```
$ make attiny85 RS485=B2 LED_START_FLASHES=0 TIMEOUT=8 BAUD_RATE=19200 NO_START_APP_ON_POR=1 START_APP_ON_EXTR=1
avr-gcc.exe (GCC) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 19200, Real: 19230, UBRRL = 51, Difference=0.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=attiny85 -DF_CPU=8000000L  -DBAUD_RATE=19200 '-DLED_START_FLASHES=0'           -DWDTTIME=8 -DNO_START_APP_ON_POR=1 -DSTART_APP_ON_EXTR=1 -DRS485=B2 '-DVIRTUAL_BOOT_PARTITION' '-DSOFT_UART'   -c -o optiboot.o optiboot.c
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=attiny85 -DF_CPU=8000000L  -DBAUD_RATE=19200 '-DLED_START_FLASHES=0'           -DWDTTIME=8 -DNO_START_APP_ON_POR=1 -DSTART_APP_ON_EXTR=1 -DRS485=B2 '-DVIRTUAL_BOOT_PARTITION' '-DSOFT_UART' -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_attiny85.elf optiboot.o
avr-size optiboot_attiny85.elf
   text    data     bss     dec     hex filename
    576       0       0     576     240 optiboot_attiny85.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_attiny85.elf optiboot_attiny85.hex
avr-objdump -h -S optiboot_attiny85.elf > optiboot_attiny85.lst
rm optiboot.o

```

mcuee, 2022-12-10:
I guess this issue can be closed.