Optiboot/optiboot #353: optiboot_x fails for >48k of flash (ie xTiny, AVRDx64, AVRDx128)

WestfW, 2022-10-11:
The current optiboot_x was written for ATmega4809, and assumes that all of flash memory is mapped into the RAM address space.  Also, it doesn't handle any of the hacks for an extended address byte.

So optiboot_x will fail to work for more than the amount of flash that is directly mapped into the RAM address space by default.  That's only 32k on the 64k and 128k devices.

https://github.com/avrdudes/avrdude/issues/1120#

@SpenceKonde @MCUdude @avrdudes

SpenceKonde, 2022-10-11:
Uh. the version that comes with DxCore most certainly works. Significant changes were required, resulting in the creatively name optiboot_dx. (the main changes were adapting to the different write method, realising that no bootloader that does not clear reset cause flags is fit for production, and did the write to flash from app TOTALLY differently.

I'm concerned by the way thing are getting changed with avrdude in ways that break optiboot. 

My plan is to continue using avrdude 6.3 in my releases until the commortion dies down.

mcuee, 2022-10-15:
I believe optiboot_dx works here with avrdude 7.0 release, even though there is a strange verification error prompt.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude7 -C .\avrdude7.conf -c arduino -P COM6 -b 115200 -p avr128db48 
-qq -D -U .\hex2\hexaa64KB_with_blink.hex && echo OK
avrdude7.exe: verification error, first mismatch at byte 0x0200
              0xff != 0x79
avrdude7.exe: verification error; content mismatch

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c pkobn_updi -p avr128db48 -qq -D -U flash:v:.\hex2\hexaa64KB_with_blink.hex:i && echo OK
                 Vtarget                      : 3.31 V
                 PDI/UPDI clock Xmega/megaAVR : 100 kHz
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude7 -C .\avrdude7.conf -c arduino -P COM6 -b 115200 
-p avr128db48 -qq -D -U flash:v:.\hex2\hexaa64KB_with_blink.hex:i && echo OK
OK

```

mcuee, 2022-10-22:
The following avrdude PR should fix the issues with optiboot_x and optiboot_dx for avrdude git main.
* https://github.com/avrdudes/avrdude/pull/1140

WestfW, 2022-11-12:
So ... it turns out that the mega0 and xTiny chips supported by Optiboot_x are substantially different than the new AVR-Dx, and it seems that "fixing" Optiboot_x (or Optiboot) is not a good strategy.

@SpenceKonde has implemented a new bootloader "Optiboot_dx", based on optiboot, that does support them as part of https://github.com/SpenceKonde/DxCore, and I suspect that will be the final state of things.

SpenceKonde, 2022-11-12:
Yeah - while Optiboot-dx was derived from Optiboot-x, the different method of writing the flash makes the whole thing a very different animal, and I don't think it makes sense to try to combine them - would just be ifdef hell. 

There are still a few unknowns about the future Ex-series, which seems mostly very similar to the tinyAVR/mega0. The main mystery is how they'll do the flash addressing, since it can have up to 64k of flash, and only memory maps half of it. That might be viable to add onto Optiboot-x, rather than having to make an Optiboot-ex.

mcuee, 2022-12-10:
In any case, optiboot_dx work with AVR Dx parts >48k flash. 

As of now, there are no >48K Flash AVR parts for mega0/1 and ATtiny0/1/2. So this is not really an issue either.

Probably this issue can be closed.