Optiboot/optiboot #360: m328pb: source compiles fine but bootloader does not work

0xCAFEAFFE, 2022-12-20:
I'll soon be tearing my hair out, I'm simply trying to compile optiboot (unmodified _for now_) from source for an ATmega328PB and can't get it to work:

1) Hardware:
The MCU is sitting on a custom PCB with a CH340 bus converter, the circuit was designed for an m88pa and previously worked well, now I need more memory, therefore the upgrade to the m328pb.
I don't see the hardware being at fault because **bootloader and application work just fine if I use a precompiled .hex file** of [optiboot_flash from MCUdude's MiniCore](https://github.com/MCUdude/MiniCore/blob/master/avr/bootloaders/optiboot_flash/bootloaders/atmega328pb/8000000L/optiboot_flash_atmega328pb_UART0_9600_8000000L_B5.hex). My fuse bits are E2 DC FF, external programmer is a USBasp.

2) Toolchain:
I'm on a Windows machine, after not getting anywhere with Microchip Studio (7.0.2542) or the makefile (not 100% independent of OS) I reverted to building manually in the commandline using avr-gcc 5.4.0 (AVR_8_bit_GNU_Toolchain_3.6.2_1778, supplied with above version of Microchip Studio).

I am using the most recent source of optiboot and run the following commands to compile, link, translate and disassemble:

`avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=8000000L -DBAUD_RATE=9600 -DLED=B5 optiboot.c -c`
`avr-gcc -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles optiboot.o -o optiboot.elf`
`avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot.elf optiboot.hex`
`avr-objdump -d optiboot.o > optiboot.asm`
`avr-size -C optiboot.elf`

The arguments I pieced together from the makefile, the commands produce the expected files, no warnings, no errors, no output except avr-size of course:

```
AVR Memory Usage
----------------
Device: Unknown

Program:     412 bytes
(.text + .data + .bootloader)

Data:          0 bytes
(.data + .bss + .noinit)
```

The disassembly and the executable are attached here, masquerading as text files as GitHub does not allow all file extensions:
[optiboot_asm.txt](https://github.com/Optiboot/optiboot/files/10271283/optiboot_asm.txt)
[optiboot_hex.txt](https://github.com/Optiboot/optiboot/files/10271284/optiboot_hex.txt)

I can of course upload the executable to my MCU with the USBasp but then the bootloader does not respond to avrdude (version 7.0 via AVRDUDESS 2.14) at all:

```
avrdude.exe: Send: 0 [30]   [20] 
avrdude.exe: Send: 0 [30]   [20] 
avrdude.exe: Send: 0 [30]   [20] 
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
```

I don't know what I am missing, unfortunately there is no precompiled .hex file for the ATmega328PB that I could compare my output to. If someone could provide a working .hex or .lst file, that might already help figuring out what's wrong.

Any input appreciated.

hecko, 2022-12-20:
Wondering whether you have tried the hex from 

https://github.com/MCUdude/MiniCore/tree/master/avr/bootloaders/optiboot_flash/bootloaders/atmega328pb

See atmega328pb_build_info.txt in that file. You probably want to try 

https://github.com/MCUdude/MiniCore/blob/master/avr/bootloaders/optiboot_flash/bootloaders/atmega328pb/8000000L/optiboot_flash_atmega328pb_UART0_9600_8000000L_B5.hex ?

Im very curious about this myself so can you please report back how this works out for you?

mcuee, 2022-12-21:
Somehow I could not get ATmega328P Xplained Mini to work with either Optiboot hex file or MiniCore hex file either.

However, the autobaud urboot firmware works fine.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb -U .\urboot_atmega328pb_autobaud_ee_lednop_fr_ce.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\urboot_atmega328pb_autobaud_ee_lednop_fr_ce.hex for flash
         with 470 bytes in 2 sections within [0x7e00, 0x7fff]
         using 4 pages and 42 pad bytes
avrdude: writing 470 bytes flash ...

Writing | ################################################## | 100% 0.13 s

avrdude: 470 bytes of flash written
avrdude: verifying flash memory against .\urboot_atmega328pb_autobaud_ee_lednop_fr_ce.hex

Reading | ################################################## | 100% 0.01 s

avrdude: 470 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM14  -p m328pb -U .\Blink.ino.atmega328pb-xmini.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino.atmega328pb-xmini.hex for flash
         with 928 bytes in 1 section within [0, 0x39f]
         using 8 pages and 96 pad bytes
avrdude: writing 928 bytes flash ...

Writing | ################################################## | 100% 0.18 s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.atmega328pb-xmini.hex

Reading | ################################################## | 100% 0.13 s

avrdude: 928 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2022-12-21:
Reference from MiniCore. It is said somehow UART0 bootloader does not work.
* https://github.com/MCUdude/MiniCore/issues/228

0xCAFEAFFE, 2022-12-21:
@hecko:
> You probably want to try
> 
> https://github.com/MCUdude/MiniCore/blob/master/avr/bootloaders/optiboot_flash/bootloaders/atmega328pb/8000000L/optiboot_flash_atmega328pb_UART0_9600_8000000L_B5.hex ?

This is the exact file I used and linked above, the one worked for me.

@mcuee:
Since the working .hex file above uses UART0, I'm not sure if this is related. Using UART1 is unfortunately off the table for me, since the circuit was designed to use UART0.

0xCAFEAFFE, 2022-12-21:
My problem seems not to be limited only to the m328pb: compiling for the m8 also gives a different result compared what's in the repo! Why are all the executables that I compile missing a few bytes? Am I missing an argument of avr-gcc?

mcuee, 2022-12-21:
You can refer to this issue on the building of optiboot under Windows. I use git bash plus the Atmel toolchain.
* https://github.com/Optiboot/optiboot/issues/355

<details>
<summary> Click to see the detailed build log </summary>

```

$ PATH=/c/work/atmel/atmel_avr8_5.4.0_win32/bin:$PATH

$ export PATH

$ avr-gcc -v
Using built-in specs.
Reading specs from c:/work/atmel/atmel_avr8_5.4.0_win32/bin/../lib/gcc/avr/5.4.0/device-specs/specs-avr2
COLLECT_GCC=C:\work\atmel\atmel_avr8_5.4.0_win32\bin\avr-gcc.exe
COLLECT_LTO_WRAPPER=c:/work/atmel/atmel_avr8_5.4.0_win32/bin/../libexec/gcc/avr/5.4.0/lto-wrapper.exe
Target: avr
Configured with: /home/toolsbuild/workspace/avr8-gnu-toolchain/src/gcc/configure LDFLAGS=-L/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86-hostlibs/lib CPPFLAGS= --target=avr --host=i686-w64-mingw32 --build=x86_64-pc-linux-gnu --prefix=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86 --libdir=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86/lib --enable-languages=c,c++ --with-dwarf2 --enable-doc --disable-shared --disable-libada --disable-libssp --disable-nls --with-avrlibc=yes --with-mpfr=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86-hostlibs --with-gmp=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86-hostlibs --with-mpc=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-win32_x86-hostlibs --enable-win32-registry=avrtoolchain --with-pkgversion=AVR_8_bit_GNU_Toolchain_3.6.2_1778 --with-bugurl=http://www.microchip.com
Thread model: single
gcc version 5.4.0 (AVR_8_bit_GNU_Toolchain_3.6.2_1778)

$ ./makeall.arduino.sh
rm -rf *.o *.elf *.lst *.map *.sym *.lss *.eep *.srec *.bin *.hex *.tmp.sh
"C:/ProgramData/chocolatey/lib/make/tools/install/bin/make.exe" atmega168 AVR_FREQ=8000000L LED_START_FLASHES=3
make[1]: Entering directory 'C:/work/avr/avrdude_test/others/optiboot/optiboot/bootloaders/optiboot'
avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1778) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 111111, UBRRL = 8, Difference=-3.5%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega168 -DF_CPU=8000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3                 -c -o optiboot.o optiboot.c
optiboot.c:422:2: warning: #warning BAUD_RATE off by greater than -2% [-Wcpp]
 #warning BAUD_RATE off by greater than -2%
  ^
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega168 -DF_CPU=8000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3               -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega168.elf optiboot.o
avr-size optiboot_atmega168.elf
   text    data     bss     dec     hex filename
    474       0       0     474     1da optiboot_atmega168.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega168.elf optiboot_atmega168.hex
avr-objdump -h -S optiboot_atmega168.elf > optiboot_atmega168.lst
rm optiboot.o
make[1]: Leaving directory 'C:/work/avr/avrdude_test/others/optiboot/optiboot/bootloaders/optiboot'
mv optiboot_atmega168.hex optiboot_lilypad.hex
mv optiboot_atmega168.lst optiboot_lilypad.lst
"C:/ProgramData/chocolatey/lib/make/tools/install/bin/make.exe" atmega168 AVR_FREQ=8000000L LED_START_FLASHES=3
make[1]: Entering directory 'C:/work/avr/avrdude_test/others/optiboot/optiboot/bootloaders/optiboot'
avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1778) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 111111, UBRRL = 8, Difference=-3.5%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega168 -DF_CPU=8000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3                 -c -o optiboot.o optiboot.c
optiboot.c:422:2: warning: #warning BAUD_RATE off by greater than -2% [-Wcpp]
 #warning BAUD_RATE off by greater than -2%
  ^
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega168 -DF_CPU=8000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3               -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega168.elf optiboot.o
avr-size optiboot_atmega168.elf
   text    data     bss     dec     hex filename
    474       0       0     474     1da optiboot_atmega168.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega168.elf optiboot_atmega168.hex
avr-objdump -h -S optiboot_atmega168.elf > optiboot_atmega168.lst
rm optiboot.o
make[1]: Leaving directory 'C:/work/avr/avrdude_test/others/optiboot/optiboot/bootloaders/optiboot'
mv optiboot_atmega168.hex optiboot_pro_8MHz.hex
mv optiboot_atmega168.lst optiboot_pro_8MHz.lst
"C:/ProgramData/chocolatey/lib/make/tools/install/bin/make.exe" atmega168 AVR_FREQ=16000000L LED_START_FLASHES=3
make[1]: Entering directory 'C:/work/avr/avrdude_test/others/optiboot/optiboot/bootloaders/optiboot'
[... 339 lines cut ...]
   text    data     bss     dec     hex filename
    474       0       0     474     1da optiboot_atmega168.elf
[... cut ...]
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3               -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o
avr-size optiboot_atmega328.elf
   text    data     bss     dec     hex filename
    474       0       0     474     1da optiboot_atmega328.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
avr-objdump -h -S optiboot_atmega328.elf > optiboot_atmega328.lst
rm optiboot.o
```

</details>

hecko, 2022-12-21:
Right, can you clarify which optiboot tag are you pulling from git, avr-gcc version (and how it was installed) and also maybe provide a avrdude commands you are using?

I can test the whole 1:1 procedure using atmel-ice over on my side and at least give you some kind of feedback.

mcuee, 2022-12-21:
> PS: Im wondering whether having something like Docker or Podman to be used for compiling Optiboot rather than figuring out the toolchain combination would be a way to go in situations like these.

Yes this is a good way to build optiboot as well as avrdude.

Just FYI, if you want to try latest snapshot of avrdude under Linux/macOS/Windows, you can try my snapshot release here, built using the Docker Container provided by Arduino.
https://github.com/mcuee/avrdude-packing

That being said, it is not that difficult to build optiboot under Windows, I am OS-neutral guy so I will always try to build things like optiboot across Linux/Windows/macOS natively.

You can refer to my issues in different optiboot related repos.

https://github.com/Optiboot/optiboot/issues/355
https://github.com/MCUdude/optiboot_flash/issues/8 
https://github.com/SpenceKonde/DxCore/issues/374 (Done)
https://github.com/SpenceKonde/megaTinyCore/issues/870
https://github.com/SpenceKonde/ATTinyCore/issues/732

hecko, 2022-12-21:
@mcuee Ill try to use the https://ww1.microchip.com/downloads/aemDocuments/documents/DEV/ProductDocuments/SoftwareTools/avr8-gnu-toolchain-3.7.0.1796-linux.any.x86_64.tar.gz to compile the optiboot for atmega328pb on Linux and then create a Dockerfile - if you have Dockerfile for this combination available I would be happy to test this.

mcuee, 2022-12-21:
> @mcuee Ill try to use the https://ww1.microchip.com/downloads/aemDocuments/documents/DEV/ProductDocuments/SoftwareTools/avr8-gnu-toolchain-3.7.0.1796-linux.any.x86_64.tar.gz to compile the optiboot [...]

As of now I am not using Docker for Optiboot myself since I have no issues building Optiboot under Linux/macOS/Windows natively. 

It will be great that you can share the Docker file here for others to use. But one suggestion, you may want to use the older version which may produce tighter codes. Or even better if you can switch between the two toolchain. The Atmel toolchain (5.4.0 or 7.3.0) does not support ATmega328PB though so you may need to copy the file from either Arduino toolchain or Microchip Studio DFPs.

https://ww1.microchip.com/downloads/aemDocuments/documents/DEV/ProductDocuments/SoftwareTools/avr8-gnu-toolchain-3.7.0.1796-linux.any.x86_64.tar.gz (newer avr-gcc 7.3.0 version).

https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/SoftwareLibraries/Firmware/avr8-gnu-toolchain-3.6.2.1778-linux.any.x86_64.tar.gz  (older avr-gcc 5.4.0 version)

Yet the other option is to use the toolchain from DxCore which adds support for building optiboot_x and optiboot_dx (version 7.3.0). It has the support for ATmega328PB already.
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azzy1-x86_64-pc-linux-gnu.tar.bz2

Optiboot code size comparison of different compilers
[image]

hecko, 2022-12-21:
@mcuee can you suggest why Im getting the following error? 

```bash
maco@lists:~/dl/optiboot-8.0/optiboot/bootloaders/optiboot$ ./testc.sh
+ GCC='../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-gcc -mmcu=atmega328pb -B ../../../../atpack/gcc/dev/atmega328pb/ -I ../../../../atpack/include/'
+ OBJCOPY=../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-objcopy
+ OBJDUMP=../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-objdump
+ SIZE=../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-size
+ ../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-gcc -mmcu=atmega328pb -B ../../../../atpack/gcc/dev/atmega328pb/ -I ../../../../atpack/include/ -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=8000000L -DBAUD_RATE=9600 -DLED=B5 optiboot.c -c
+ ../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-gcc -mmcu=atmega328pb -B ../../../../atpack/gcc/dev/atmega328pb/ -I ../../../../atpack/include/ -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles optiboot.o -o optiboot.elf
/home/maco/dl/avr8-gnu-toolchain-linux_x86_64/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot open linker script file link_optiboot.ld: No such file or directory
collect2: error: ld returned 1 exit status
```

mcuee, 2022-12-21:
> My fuse bits are E2 DC FF
@0xCAFEAFFE 
Going back to your issue, I think your fuse setting is not correct. `hfuse` should be `0xDE` and not `0xDC` as the bootloader size unit is `WORD` and not `BYTE`.

0xCAFEAFFE, 2022-12-21:
> Right, can you clarify which optiboot tag are you pulling from git also maybe provide a avrdude commands you are using?

I tried optiboot-master and the optiboot-8.0 tag, both result in slightly different executables of the same size, both don't work.

avrdude arguments to flash bootloader via USBasp:
`-c usbasp -p m328pb -P usb -v -e -U flash:w:"optiboot.hex":i `

avrdude arguments to upload using optiboot:
`-c arduino -p m328pb -P COM3 -b 9600 -v -D -U flash:w:"blinky.hex":i `

0xCAFEAFFE, 2022-12-21:
> > My fuse bits are E2 DC FF
> 
> @0xCAFEAFFE Going back to your issue, I think your fuse setting is not correct. `hfuse` should be `0xDE` and not `0xDC` as the bootloader size unit is `WORD` and not `BYTE`.

Ohh, that's interesting, thank you for pointing that out! Curious, that the precompiled .hex still worked..
I tried the correct setting now, does not fix it though..

mcuee, 2022-12-21:
> @mcuee can you suggest why Im getting the following error?
> 
> ```shell
> /home/maco/dl/avr8-gnu-toolchain-linux_x86_64/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot open linker script file link_optiboot.ld: No such file or directory
> collect2: error: ld returned 1 exit status
> ```

I thinxisk this is because the file `link_optiboot.ld` does not exist in the repo.

mcuee, 2022-12-21:
This is what I get under Linux.
```
mcuee@UbuntuSwift3:~/build/avr/optiboot/optiboot/bootloaders/optiboot$ make atmega328pb AVR_FREQ=8000000L BAUD_RATE=57600 LED_START_FLASHES=3 LED=B5
avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 57600, Real: 58823, UBRRL = 16, Difference=2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=8000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3    -DLED=B5              -c -o optiboot.o optiboot.c
optiboot.c:425:2: warning: #warning BAUD_RATE off by greater than 2% [-Wcpp]
 #warning BAUD_RATE off by greater than 2%
  ^~~~~~~
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=8000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3    -DLED=B5            -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328pb_UART_57600_8000000L.elf optiboot.o 
avr-size optiboot_atmega328pb_UART_57600_8000000L.elf
   text	   data	    bss	    dec	    hex	filename
    496	      0	      0	    496	    1f0	optiboot_atmega328pb_UART_57600_8000000L.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328pb_UART_57600_8000000L.elf optiboot_atmega328pb_UART_57600_8000000L.hex
avr-objdump -h -S optiboot_atmega328pb_UART_57600_8000000L.elf > optiboot_atmega328pb_UART_57600_8000000L.lst
rm optiboot.o

```

mcuee, 2022-12-21:
My board is using 16MHz external clock.
I can confirm that optiboot works fine.

```
$ make atmega328pb AVR_FREQ=16000000L BAUD_RATE=57600 LED_START_FLASHES=3 LED=B5
avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 57600, Real: 57142, UBRRL = 34, Difference=0.7%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3    -DLED=B5              -c -o optiboot.o optiboot.c
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3    -DLED=B5            -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328pb_UART_57600_16000000L.elf optiboot.o
avr-size optiboot_atmega328pb_UART_57600_16000000L.elf
   text    data     bss     dec     hex filename
    490       0       0     490     1ea optiboot_atmega328pb_UART_57600_16000000L.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328pb_UART_57600_16000000L.elf optiboot_atmega328pb_UART_57600_16000000L.hex
avr-objdump -h -S optiboot_atmega328pb_UART_57600_16000000L.elf > optiboot_atmega328pb_UART_57600_16000000L.lst
rm optiboot.o

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb
 -U .\optiboot_atmega328pb_UART_57600_16000000L.hex -qq && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM14 -b 57600 -p m328pb  -U .\Blink.ino.atmega328pb-xmini.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino.atmega328pb-xmini.hex for flash
         with 928 bytes in 1 section within [0, 0x39f]
         using 8 pages and 96 pad bytes
avrdude: writing 928 bytes flash ...

Writing | ################################################## | 100% 0.31 s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.atmega328pb-xmini.hex

Reading | ################################################## | 100% 0.23 s

avrdude: 928 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2022-12-21:
It does not work with 115200bps but it does work at 125000bps.
```
$ make atmega328pb AVR_FREQ=16000000L BAUD_RATE=125000 LED_START_FLASHES=3 LED=B5
avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 125000, Real: 125000, UBRRL = 15, Difference=0.0%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=16000000L  -DBAUD_RATE=125000 -DLED_START_FLASHES=3    -DLED=B5              -c -o optiboot.o optiboot.c
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=16000000L  -DBAUD_RATE=125000 -DLED_START_FLASHES=3    -DLED=B5            -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328pb_UART_125000_16000000L.elf optiboot.o
avr-size optiboot_atmega328pb_UART_125000_16000000L.elf
   text    data     bss     dec     hex filename
    490       0       0     490     1ea optiboot_atmega328pb_UART_125000_16000000L.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328pb_UART_125000_16000000L.elf optiboot_atmega328pb_UART_125000_16000000L.hex
avr-objdump -h -S optiboot_atmega328pb_UART_125000_16000000L.elf > optiboot_atmega328pb_UART_125000_16000000L.lst
rm optiboot.o

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb -U .\optiboot_atmega328pb_UART_125000_16000000L.hex -qq && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM14 -b 125000 -p m328pb  -U .\Blink.ino.atmega328pb-xmini.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino.atmega328pb-xmini.hex for flash
         with 928 bytes in 1 section within [0, 0x39f]
         using 8 pages and 96 pad bytes
avrdude: writing 928 bytes flash ...

Writing | ################################################## | 100% 0.16 s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.atmega328pb-xmini.hex

Reading | ################################################## | 100% 0.12 s

avrdude: 928 bytes of flash verified

avrdude done.  Thank you.

```

hecko, 2022-12-21:
So I have tried to re-run @mcuee compilation process by running: 

```bash
#!/bin/bash

set -e
set -x

GCC="../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-gcc -B ../../../../atpack/gcc/dev/atmega328pb/ -I ../../../../atpack/include/"
OBJCOPY="../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-objcopy"
OBJDUMP="../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-objdump"
SIZE="../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-size"

$GCC -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=8000000L -DBAUD_RATE=9600 -DLED=B5 -DLED_START_FLASHES=3 -c -o optiboot.o optiboot.c
#$GCC -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles optiboot.o -o optiboot.elf
$GCC -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=8000000L -DBAUD_RATE=9600 -DLED=B5 -DLED_START_FLASHES=3 -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot.elf optiboot.o

$OBJCOPY -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot.elf optiboot.hex
$OBJDUMP -h -S optiboot.elf > optiboot.lst
$SIZE -C optiboot.elf
```

And the result: 
```bash
maco@lists:~/dl/optiboot-8.0/optiboot/bootloaders/optiboot$ ./testc.sh
+ GCC='../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-gcc -B ../../../../atpack/gcc/dev/atmega328pb/ -I ../../../../atpack/include/'
+ OBJCOPY=../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-objcopy
+ OBJDUMP=../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-objdump
+ SIZE=../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-size
+ ../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-gcc -B ../../../../atpack/gcc/dev/atmega328pb/ -I ../../../../atpack/include/ -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=8000000L -DBAUD_RATE=9600 -DLED=B5 -DLED_START_FLASHES=3 -c -o optiboot.o optiboot.c
+ ../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-gcc -B ../../../../atpack/gcc/dev/atmega328pb/ -I ../../../../atpack/include/ -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=8000000L -DBAUD_RATE=9600 -DLED=B5 -DLED_START_FLASHES=3 -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot.elf optiboot.o
+ ../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot.elf optiboot.hex
+ ../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-objdump -h -S optiboot.elf
+ ../../../../avr8-gnu-toolchain-linux_x86_64/bin/avr-size -C optiboot.elf
AVR Memory Usage
----------------
Device: Unknown

Program:     494 bytes
(.text + .data + .bootloader)

Data:          0 bytes
(.data + .bss + .noinit)

maco@lists:~/dl/optiboot-8.0/optiboot/bootloaders/optiboot$ cat optiboot.hex
:107E000001C0E5C0112484B7882361F0982F9A70CF
:107E1000923041F081FF02C097EF94BF282E80E09E
:107E2000C1D0EEC085E08093810082E08093C000E5
:107E300088E18093C10086E08093C20087E680934A
:107E4000C4008EE0AFD0259A86E028E13EEF91E0B5
:107E5000309385002093840096BBB09BFECF1D9A83
:107E6000A8954091C00047FD02C0815089F793E07A
:107E7000E92EDD24D39425E0C22E31E1B32E86D045
:107E8000813479F483D0182F93D0123811F480E024
:107E900004C088E0113809F083E071D080E16FD030
:107EA000EECF823419F484E18BD0F8CF853411F40D
:107EB00085E0FACF853541F469D0C82F67D0D82F37
:107EC000CC0FDD1F75D0EACF863519F484E078D069
:107ED000DECF843609F034C059D058D0F82E56D0B1
:107EE000A82E00E011E048018FEF881A980A4ED0C2
:107EF000F80180838401F810F6CF5AD0F5E4AF1270
:107F000001C0FFCFFE01E7BEE89507B600FCFDCF3C
:107F1000FE01A0E0B1E0CD0102962D913C91090156
:107F2000D7BEE89511243296DC01F812F4CFFE0199
:107F3000C7BEE89507B600FCFDCFB7BEE895AECF4B
:107F4000843771F423D022D0F82E20D031D08E0186
:107F5000F80185918F0113D0FA94F110F9CF9ECFDB
:107F6000853739F425D08EE10AD085E908D086E13D
:107F700094CF813509F0A6CF88E014D0A3CF90919B
:107F8000C00095FFFCCF8093C60008958091C0008B
:107F900087FFFCCF8091C00084FD01C0A89580912F
:107FA000C6000895E0E6F0E098E1908380830895AC
:107FB000EDDF803219F088E0F5DFFFCF84E1DFCF1D
:107FC000CF93C82FE3DFC150E9F7CF91F1CFFC0188
:107FD0000A0167BFE895112407B600FCFDCF667063
:0E7FE00029F0452B19F481E187BFE89508953B
:027FFE00000879
:0400000300007E007B
:00000001FF
```

The diff in HEX files from the first post from @0xCAFEAFFE is very different. Its very close to the HEX from https://github.com/Optiboot/optiboot/issues/360#issuecomment-1361261237

hecko, 2022-12-21:
@mcuee the Dockerfile (and docs) used to generate the above optiboot.hex (1:1) for Atmega328PB, 8MHz, 9600, B5 available here: https://github.com/hecko/optiboot_docker_build

@0xCAFEAFFE would this help you? Can you test the generated hex file wheter this works for you?

0xCAFEAFFE, 2022-12-21:
Thank's @hecko, that does indeed help, your .hex file is working for me with the internal 8MHz RC osc!
Could you please also send me a .lst file and/or a disassembly listing generated with avr-objdump -d?

The .hex file from mcuee does not quite work (optiboot answers but avrdude reads back garbage) because the baud rate of 57600bd is not compatible with the 8MHz clock.

hecko, 2022-12-21:
Attached - please let us know how you get on with this.
[optiboot_asm.txt](https://github.com/Optiboot/optiboot/files/10279120/optiboot_asm.txt)
[optiboot_hex.txt](https://github.com/Optiboot/optiboot/files/10279121/optiboot_hex.txt)
[optiboot_lst.txt](https://github.com/Optiboot/optiboot/files/10279122/optiboot_lst.txt)

0xCAFEAFFE, 2022-12-21:
Oh dear.. I found my mistake and I am embarrassed to admit it: I made a "temporary" change to my sfr_defs.h a while ago and forgot about it in the meantime.. Thanks to everyone!

closed 2022-12-21 by 0xCAFEAFFE

SpenceKonde, 2023-06-14:
> It does not work with 115200bps but it does work at 125000bps (and even 250000bps).

Is the bootloader using U2X? It. The baud rate generator on classic AVRs was shit, that's why some cores support UART clock crystals, because that was the only way to get decent calculation accuracy . 

115200 is also a *particularly* bad speed for 16 MHz classic AVR.

76800 (2/3rds of 115200 baud would work much better at 16 MHz, though for 12 MHz or for 16.5 MHz, 115200 baud is better.... 

https://docs.google.com/spreadsheets/d/1uzU_HqWEpK-wQUo4Q7FBZGtHOo_eY4P_BAzmVbKCj6Y/edit?usp=sharing