Optiboot/optiboot #364: Incomplete optiboot_x build for mega0 and ATtiny0/1/2 AVRs

mcuee, 2022-12-22:
The following is not complete, missing many chips.
https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/makeall.mega0.sh

mcuee, 2022-12-22:
Reference:
https://github.com/MCUdude/MegaCoreX (all chips will support optiboot_x)
https://github.com/SpenceKonde/megaTinyCore (most of the chip will support optiboot_x)

WestfW, 2022-12-22:
But many of those chips will wind up with identical code paths and binarie, since (for example) tiny0/tiny1/tiny2 are identical wrt the bootloader, and flash size is no longer relevant.  The makeall.mega0 script should be testing at least one rom each relevant part of the tree.

im leaving it up to Spence and mcudude to build the exhaustive list of .hex files, if they desire.
(Or that’s the theory anyway.  I could be convinced that I’ve left out important examples.)

mcuee, 2022-12-22:
> But many of those chips will wind up with identical code paths and binarie, since (for example) tiny0/tiny1/tiny2 are identical wrt the bootloader, and flash size is no longer relevant. The makeall.mega0 script should be testing at least one rom each relevant part of the tree.

That is indeed true.
Reference: https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/bootloaders/hex

> im leaving it up to Spence and mcudude to build the exhaustive list of .hex files, if they desire.
> (Or that’s the theory anyway. I could be convinced that I’ve left out important examples.)

Fair enough.

mcuee, 2022-12-22:
In that case, I will close this issue for now.

closed 2022-12-22 by mcuee