Optiboot/optiboot #370: unable to build

quantrpeter, 2023-07-12:
hi
    build has error but still producing the hex file, after i burn it to arduino nano, not working. The build error is:

```
C:\Users\Peter\Desktop\optiboot\optiboot\bootloaders\optiboot\Makefile(682,1): error: recipe for target 'baudcheck' failed
		process_begin: CreateProcess(NULL, bash baudcheck.tmp.sh, ...) failed.
		make (e=2): The system cannot find the file specified.
		make[1]: [baudcheck] Error 2 (ignored)
		avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3                 -c -o optiboot.o optiboot.c
		avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3               -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o 
		avr-size optiboot_atmega328.elf
		   text	   data	    bss	    dec	    hex	filename
		    474	      0	      0	    474	    1da	optiboot_atmega328.elf
		avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
		avr-objdump -h -S optiboot_atmega328.elf > optiboot_atmega328.lst
		rm optiboot.o
```

[image]

thanks

WestfW, 2023-07-15:
The baudcheck step of make relies on some utilities that may or may not be present on windows systems.  It's mostly pretty-printing of baud rate issues, and the build will work fine without it.

I'll try to do some checking on exactly which packages result in having the full util set.

BerranRemzi, 2025-11-06:
It's working fine with **Git Bash**.
I followed these steps to compile:
1. Open https://github.com/Optiboot/Toolchains
2. Download [Tools for Windows.](https://drive.google.com/file/d/1CFvsTWSxPleEXqDdv3nCtOqSoFpY7NCP/view?usp=sharing)
3. Extract AVRTOOLS-windows.zip to C:\Tools
4. Open `optiboot/bootloaders/optiboot` with **Git Bash**
5. Run:
`export PATH=$PATH:"/C/Tools/AVRTOOLS/bin"`
6. Finally, compile using one of the following commands:
`make atmega328`
or for a smaller binary:
`make LED_START_ON=1 LED_START_FLASHES=0 atmega328`