Optiboot/optiboot #380: EEPROM writing fails

owenduffy, 2024-06-29:
I have used a custom build of an earlier version of Optiboot for atmega328, and I could read and write EEPROM using AVRDUDE.

I have created a build using the current version of Optiboot:

`D:\src\optiboot\optiboot\bootloaders\optiboot>make atmega328 LED_START_FLASHES=0  SUPPORT_EEPROM=1
avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -g -Wall -Os
-fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=0      -DSUPPOR
T_EEPROM          -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size optiboot_atm
ega328.elf
   text    data     bss     dec     hex filename
    490       0       0     490     1ea optiboot_atmega328.elf
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -j .text
-j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objdump -h -S opt
iboot_atmega328.elf > optiboot_atmega328.lst
`
Seems good, so I installed it with an ISP programmer.

`>>>: avrdude.exe -c usbasp -p m328p -P usb -b 115200 -B 8.0 -v -U flash:w:"D:\src\optiboot\optiboot\bootloaders\optiboot\optiboot_atmega328.hex":i 

avrdude: Version 7.3
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is D:\Program Files (x86)\AVRDUDESS\avrdude.conf

         Using port            : usb
         Using programmer      : usbasp
         Setting baud rate     : 115200
         Setting bit clk period: 8.0 us
         AVR Part              : ATmega328P
         Programming modes     : ISP, HVPP, debugWIRE, SPM
         Programmer Type       : usbasp
         Description           : USBasp ISP and TPI programmer
avrdude: set SCK frequency to 93750 Hz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: set SCK frequency to 93750 Hz

avrdude: processing -U flash:w:D:\src\optiboot\optiboot\bootloaders\optiboot\optiboot_atmega328.hex:i
avrdude: reading input file D:\src\optiboot\optiboot\bootloaders\optiboot\optiboot_atmega328.hex for flash
         with 490 bytes in 2 sections within [0x7e00, 0x7fff]
         using 4 pages and 22 pad bytes
avrdude: writing 490 bytes flash ...
Writing | ################################################## | 100% 0.07s
avrdude: 490 bytes of flash written
avrdude: verifying flash memory against D:\src\optiboot\optiboot\bootloaders\optiboot\optiboot_atmega328.hex
Reading | ################################################## | 100% 0.00s
avrdude: 490 bytes of flash verified

avrdude done.  Thank you.`

It looks good.

So I test a read EEPROM:

`>>>: avrdude.exe -c arduino -p m328p -P COM5 -b 115200 -B 0.5 -v -U eeprom:r:"D:\src\ddwm\ddwm\eeprom\stuff.bin":r 

avrdude: Version 7.3
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is D:\Program Files (x86)\AVRDUDESS\avrdude.conf

         Using port            : COM5
         Using programmer      : arduino
         Setting baud rate     : 115200
         Setting bit clk period: 0.5 us
         AVR Part              : ATmega328P
         Programming modes     : ISP, HVPP, debugWIRE, SPM
         Programmer Type       : Arduino
         Description           : Arduino for bootloader using STK500 v1 protocol
         HW Version            : 3
         FW Version            : 8.3
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude: processing -U eeprom:r:D:\src\ddwm\ddwm\eeprom\stuff.bin:r
avrdude: reading eeprom memory ...
Reading | ################################################## | 100% 1.29s
avrdude: writing output file D:\src\ddwm\ddwm\eeprom\stuff.bin

avrdude done.  Thank you.`

Looks good.

So. let's try to write EEPROM:

`>>>: avrdude.exe -c arduino -p m328p -P COM5 -b 115200 -B 0.5 -v -U eeprom:w:"D:\src\ddwm\ddwm\eeprom\stuff.bin":r 

avrdude: Version 7.3
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is D:\Program Files (x86)\AVRDUDESS\avrdude.conf

         Using port            : COM5
         Using programmer      : arduino
         Setting baud rate     : 115200
         Setting bit clk period: 0.5 us
         AVR Part              : ATmega328P
         Programming modes     : ISP, HVPP, debugWIRE, SPM
         Programmer Type       : Arduino
         Description           : Arduino for bootloader using STK500 v1 protocol
         HW Version            : 3
         FW Version            : 8.3
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude: processing -U eeprom:w:D:\src\ddwm\ddwm\eeprom\stuff.bin:r
avrdude: reading input file D:\src\ddwm\ddwm\eeprom\stuff.bin for eeprom
         with 1024 bytes in 1 section within [0, 0x3ff]
         using 256 pages and 0 pad bytes
avrdude: writing 1024 bytes eeprom ...
Writing | -------------------------------------------------- | 0% 5.01s
avrdude stk500_recv() error: programmer is not responding
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;`

I had to kill AVRDUDE.

Did I get something wrong, or is there a bug?

Owen

WestfW, 2024-06-30:
Weird.  My compile with those options claims it doesn't fit in 512bytes...
I see the link command in your log, but not the compile itself.  Perhaps there was a stale .o file?

owenduffy, 2024-06-30:
I deleted optiboot.o and all other files produced by the build each time I rebuilt it to test.

I did try it also with BIGBOOT which was required in the 8.0 build.

Here is the build log:

`D:\src\optiboot\optiboot\bootloaders\optiboot>make atmega328 LED_START_FLASHES=0  SUPPORT_EEPROM=1
avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -g -Wall -Os
-fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=0      -DSUPPOR
T_EEPROM          -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size optiboot_atm
ega328.elf
   text    data     bss     dec     hex filename
    490       0       0     490     1ea optiboot_atmega328.elf
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -j .text
-j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objdump -h -S opt
iboot_atmega328.elf > optiboot_atmega328.lst

D:\src\optiboot\optiboot\bootloaders\optiboot>`

owenduffy, 2024-06-30:
A build with BIGBOOT:

D:\src\optiboot\optiboot\bootloaders\optiboot>make atmega328 LED_START_FLASHES=0  SUPPORT_EEPROM=1 BIGBOOT=1
avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -g -Wall -Os
-fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=0 -DBIGBOOT=1
   -DSUPPORT_EEPROM          -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size optiboot_atm
ega328.elf
   text    data     bss     dec     hex filename
    490       0       0     490     1ea optiboot_atmega328.elf
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -j .text
-j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objdump -h -S opt
iboot_atmega328.elf > optiboot_atmega328.lst

D:\src\optiboot\optiboot\bootloaders\optiboot>

WestfW, 2024-06-30:
Where exactly did you get the "most recent version of optiboot" from?
The one here puts "extra" info in flash with BIGBOOT, so there's no way it should end up with a 490byte elf file:
```
make atmega328 LED_START_FLASHES=0 SUPPORT_EEPROM=1 BIGBOOT=1
avr-gcc (AVR_8_bit_GNU_Toolchain_3.7.0_518) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=0 -DBIGBOOT=1     -DSUPPORT_EEPROM            -c -o optiboot.o optiboot.c
optiboot.c:429:2: warning: #warning BAUD_RATE off by greater than 2% [-Wcpp]
 #warning BAUD_RATE off by greater than 2%
  ^~~~~~~
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=0 -DBIGBOOT=1     -DSUPPORT_EEPROM          -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o 
avr-size optiboot_atmega328.elf
   text    data     bss     dec     hex filename
    678       0       0     678     2a6 optiboot_atmega328.elf
```

owenduffy, 2024-06-30:
> Where exactly did you get the "most recent version of optiboot" from?

Cloned from https://github.com/Optiboot/optiboot/commits/master/ a week ago.

I altered the Makefile to set the path to the tools.

 endif
 
+TOOLROOT=C:/Users/owen/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7
+GCCROOT = $(TOOLROOT)/bin/
+
 STK500 = "C:\Program Files\Atmel\AVR Tools\STK500\Stk500.exe"

owenduffy, 2024-06-30:
Ok, I changed it to point to the Studio toolchain:

D:\src\optiboot\optiboot\bootloaders\optiboot>make atmega328 LED_START_FLASHES=0  SUPPORT_EEPROM=1 BIGBOOT=1
"D:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain/bin/avr-gcc" -g -Wall -Os -fno-split-wide-typ
es -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=0 -DBIGBOOT=1     -DSUPPORT_EEPROM
            -c -o optiboot.o optiboot.c
optiboot.c:429:2: warning: #warning BAUD_RATE off by greater than 2% [-Wcpp]
 #warning BAUD_RATE off by greater than 2%
  ^
avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1778) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
"D:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain/bin/avr-gcc" -g -Wall -Os -fno-split-wide-typ
es -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=0 -DBIGBOOT=1     -DSUPPORT_EEPROM
          -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o
"D:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain/bin/avr-size" optiboot_atmega328.elf
   text    data     bss     dec     hex filename
    662       0       0     662     296 optiboot_atmega328.elf
"D:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain/bin/avr-objcopy" -j .text -j .data -j .versio
n --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
"D:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain/bin/avr-objdump" -h -S optiboot_atmega328.elf
 > optiboot_atmega328.lst

D:\src\optiboot\optiboot\bootloaders\optiboot>

I will test the write.

Owen

owenduffy, 2024-06-30:
> I will test the write.

Ok, that seems to work.

Thanks for your help.

Owen

>>>: avrdude.exe -c arduino -p m328p -P COM5 -b 115200 -B 0.5 -v -U eeprom:w:"D:\src\ddwm\ddwm\eeprom\stuff.bin":r 

avrdude: Version 7.3
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is D:\Program Files (x86)\AVRDUDESS\avrdude.conf

         Using port            : COM5
         Using programmer      : arduino
         Setting baud rate     : 115200
         Setting bit clk period: 0.5 us
         AVR Part              : ATmega328P
         Programming modes     : ISP, HVPP, debugWIRE, SPM
         Programmer Type       : Arduino
         Description           : Arduino for bootloader using STK500 v1 protocol
         HW Version            : 3
         FW Version            : 8.3
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude: processing -U eeprom:w:D:\src\ddwm\ddwm\eeprom\stuff.bin:r
avrdude: reading input file D:\src\ddwm\ddwm\eeprom\stuff.bin for eeprom
         with 1024 bytes in 1 section within [0, 0x3ff]
         using 256 pages and 0 pad bytes
avrdude: writing 1024 bytes eeprom ...
Writing | ################################################## | 100% 3.91s
avrdude: 1024 bytes of eeprom written
avrdude: verifying eeprom memory against D:\src\ddwm\ddwm\eeprom\stuff.bin
Reading | ################################################## | 100% 1.27s
avrdude: 1024 bytes of eeprom verified

avrdude done.  Thank you.

owenduffy, 2024-06-30:
A few Makefile  changes to use the Studio toolchain:

@@ -112,10 +112,13 @@ AVRDUDE_CONF = -C$(TOOLROOT)/avr/etc/avrdude.conf
 else
 GCCROOT =
 AVRDUDE_CONF =
 endif
 
+TOOLROOT=D:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain
+GCCROOT = $(TOOLROOT)/bin/
+
 STK500 = "C:\Program Files\Atmel\AVR Tools\STK500\Stk500.exe"
 STK500-1 = $(STK500) -e -d$(MCU_TARGET) -pf -vf -if$(PROGRAM)_$(TARGET).hex \
            -lFF -LFF -f$(HFUSE)$(LFUSE) -EF8 -ms -q -cUSB -I200kHz -s -wt
 STK500-2 = $(STK500) -d$(MCU_TARGET) -ms -q -lCF -LCF -cUSB -I200kHz -s -wt
 #
@@ -125,24 +128,24 @@ STK500-2 = $(STK500) -d$(MCU_TARGET) -ms -q -lCF -LCF -cUSB -I200kHz -s -wt
 OBJ        = $(PROGRAM).o
 OPTIMIZE = -Os -fno-split-wide-types -mrelax
 
 DEFS       = 
 
-CC         = $(GCCROOT)avr-gcc
+CC         = "$(GCCROOT)avr-gcc"
 
 # Override is only needed by avr-lib build system.
 
 HELPTEXT += "Option AVR_FREQ=<n>          - Clock rate of AVR CPU\n"
 
 
 override CFLAGS        = -g -Wall $(OPTIMIZE) -mmcu=$(MCU_TARGET) -DF_CPU=$(AVR_FREQ) $(DEFS)
 override LDFLAGS       = $(LDSECTIONS) -Wl,--relax -nostartfiles
 
-OBJCOPY        = $(GCCROOT)avr-objcopy
-OBJDUMP        = $(call fixpath,$(GCCROOT)avr-objdump)
+OBJCOPY        = "$(GCCROOT)avr-objcopy"
+OBJDUMP        = $(call fixpath,"$(GCCROOT)avr-objdump")
 
-SIZE           = $(GCCROOT)avr-size
+SIZE           = "$(GCCROOT)avr-size"
 
 #
 # Make command-line Options.
 # Permit commands like "make atmega328 LED_START_FLASHES=10" to pass the
 # appropriate parameters ("-DLED_START_FLASHES=10") to gcc

Thanks Bill... Owen

WestfW, 2024-06-30:
Did you figure out at all why the Arduino toolchain was behaving so strangely?  Slight changes in file size from version to version are not unexpected, but leaving out the "extra data" in bigboot is very weird...
(Example "extra info"):
```
WWHackintosh<4584> avr-objcopy -I ihex -O binary optiboot_atmega328.hex foo.bin
WWHackintosh<4585> strings foo.bin
Version=8.3
Device=atmega328p
F_CPU=16000000L
BIGBOOT=1
Built:Jun 30 2024:01:38:03
UART=0
BAUD_RATE=115200
SUPPORT_EEPROM=1
LED=B5
```

> Makefile changes to use the Studio toolchain

To use the studio toolchain, I just run "Microchip Studio Command Prompt" from the Start menu...
It adds the studio toolchain (and "shell utils" stuff) to the path for just that instance of CMD, and obviates the need for modifying the makefile...

owenduffy, 2024-06-30:
> To use the studio toolchain, I just run "Microchip Studio Command Prompt" from the Start menu...

I tried that...

`D:\src\optiboot\optiboot\bootloaders\optiboot>make atmega328 LED_START_FLASHES=0  SUPPORT_EEPROM=1 BIGBOOT=1
avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1778) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FL
ASHES=0 -DBIGBOOT=1     -DSUPPORT_EEPROM          -Wl,-Tlink_optiboot.ld -Wl,--relax -nostartfiles -o optiboot_atmega328
.elf optiboot.o
avr-size optiboot_atmega328.elf
   text    data     bss     dec     hex filename
    725       0       0     725     2d5 optiboot_atmega328.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboo
t_atmega328.hex
avr-objdump -h -S optiboot_atmega328.elf > optiboot_atmega328.lst

D:\src\optiboot\optiboot\bootloaders\optiboot>`

I note the size is different again.

That is just too easy! Thanks, I will write a note to do it this way in future.

Owen

WestfW, 2024-06-30:
(also, I think your Studio could use an update.  I'm pretty sure the current compiler is based on gcc 7.3)

owenduffy, 2024-06-30:
> (also, I think your Studio could use an update. I'm pretty sure the current compiler is based on gcc 7.3)

Thanks... I checked it before doing the tests, and it appears up to date.

Thanks again for your work on this Bill... Owen.

closed 2024-07-03 by owenduffy